powsybl/powsybl-core #121: DataSourceUtil extension handling

sylvlecl, 2017-11-24:
The handling of file extensions in DataSources is not satisfying today for several reasons.
The current mechanism, based on the creation of a DataSource based on a file name:

1/ cannot handle successive compression extensions (.gz.zip ...), although it makes not much sense...

2/ does not take into account the extension defined at the time of creation, but only at the time of writing the output

3/ deletes everything after the first dot in the name when creating the DataSource, even though it may not be part of the actual extension. It makes it impossible, for instance to read from a file "toto.tata.xiidm.gz" passed to the command line

**The DataSourceUtil and DataSource implementations need to be refactored to cope with those problems.**

mathbagu, 2018-02-07:
4/ if a folder contains two files toto.uct and toto.UCT with different contents, it's impossible to load toto.UCT because the UcteImporter will always load the first matching extension. This assessment is true with all the importers which support several extensions.

mathbagu, 2018-12-07:
This bug will be solved with the datasource refactoring #471

alicecaron, 2025-11-06:
Obsolete since the DataSource refacto