powsybl/powsybl-core #440: java.lang.NoClassDefFoundError:   javax/xml/bind/JAXBException

SanPen, 2018-09-30:
Hi, 

I am trying to compile with `./install-sh`

and I get a compilation error saying `java.lang.NoClassDefFoundError:   javax/xml/bind/JAXBException`
The stack overflow related thread is [here](https://stackoverflow.com/questions/43574426/how-to-resolve-java-lang-noclassdeffounderror-javax-xml-bind-jaxbexception-in-j)

Apparently java-xml-bind (and the java commons in general) are no longer included so the extra dependencies should be added to the download script:

```
<!-- Java 6 = JAX-B Version 2.0   -->
<!-- Java 7 = JAX-B Version 2.2.3 -->
<!-- Java 8 = JAX-B Version 2.2.8 -->
<dependency>
    <groupId>javax.xml.bind</groupId>
    <artifactId>jaxb-api</artifactId>
    <version>2.2.11</version>
</dependency>
<dependency>
    <groupId>com.sun.xml.bind</groupId>
    <artifactId>jaxb-core</artifactId>
    <version>2.2.11</version>
</dependency>
<dependency>
    <groupId>com.sun.xml.bind</groupId>
    <artifactId>jaxb-impl</artifactId>
    <version>2.2.11</version>
</dependency>
<dependency>
    <groupId>javax.activation</groupId>
    <artifactId>activation</artifactId>
    <version>1.1.1</version>
</dependency>
```
This is from Stack overflow...

flexo90, 2018-09-30:
Hello,
I had the same Problem when trying to install yesterday. There might be some other packages needed too. (i think zlib1g-dev bzip2 libbz2-dev).
I tried to document everything I did, so i would be able to quickly set it up in the future. I got everything to install after running the following commands (also ubuntu - clean install in virtual machine):

add-apt-repository ppa:webupd8team/java
apt update
apt-get install oracle-java8-installer
apt update
apt-get install oracle-java8-installer oracle-java8-set-default openjdk-8-jdk
apt-get install openjfx
apt-get install maven
apt-get install openmpi openmpi-bin openmpi-doc libopenmpi-dev
apt-get install zlib zlib-devel bzip zlib1g-dev bzip2 libbz2-dev python-qrtools

SanPen, 2018-09-30:
Thanks!

I believe that setting java-8 as the default java was the solution.

geofjamg, 2018-10-01:
Hi,

Indead, only java 8 is supported. We plan to move to Java 11 in the next days or weeks. A PR #424 for code adaptation to Java 11 is already opened. Everything is ok except a maven plugin (maven-plugin-plugin) which is not yet compatible with Java 11 (https://github.com/apache/maven-plugin-tools/pull/14). As soon as this plugin is fixed we will switch to Java 11.

SanPen, 2018-10-01:
Hi, 

A fair warning about java 11; The compiled by Oracle might ask for royalties. 

The new terms and conditions include the following gem:
> You may not: use the Programs for any data processing or any commercial, production, or internal business purposes other than developing, testing, prototyping, and demonstrating your Application;

However OpenJDK / OpenJRE is still free. See [this blog](https://blog.joda.org/2018/09/do-not-fall-into-oracles-java-11-trap.html).

closed 2023-10-11 by geofjamg