powsybl/powsybl-core #441: Intro / Documentation for people new to powsybl

flexo90, 2018-09-30:
Hello and sry for submitting an issue here :)
maybe there is something I have overlooked so far, but googling and searching through some older RTE and iTesla Presentations has not helped me in this regard. Therefore this was my only means to get more information on powsybl.

Is there a form of documentation on how to use powsybl classes or powsybl-gse? I.E. what does the class hierarchy look like. How are interfaces defined and which classes already exist. 
I managed to get everything to install and compile but now i am stuck on the next steps. I would like to try out a simple uct import, and do some basic loadflow calculations - contingency analysis, either in my own java project (preferably not) or in powsybl-gse or a command line interface. 

Best regards,
Felix

mathbagu, 2018-10-01:
Hi,
A beta version of the documentation is available in a [separated branch](https://github.com/powsybl/powsybl-core/tree/docs/docs). It's not complete yet but you can already information about:
- [iTools](https://github.com/powsybl/powsybl-core/tree/docs/docs/tools), the powsybl command line interface
- how to [configure](https://github.com/powsybl/powsybl-core/tree/docs/docs/configuration) everything
- few [tutorials](https://github.com/powsybl/powsybl-core/tree/docs/docs/tutorials) to learn how to extend powsybl features

In the next weeks, we have to write more documentation about the [different concepts](https://github.com/powsybl/powsybl-core/tree/docs/docs/architecture) and more developer tutorials.

mathbagu, 2018-12-07:
We decide to move the documentation to [powsybl](http://www.powsybl.com) website. Almost all the documentation that was in the dedicated `docs` has been moved and completed.

joesan, 2019-02-14:
Is there a Slack channel for getting in touch with you guys? I would like to contribute to the project as I got interested in the Linix Energy Foundation and came across this project. I have been working in the Utilities inidustry in Europe and would like to know more about the goals of this project so that I can participate and make some valuable contributions! Please let me know how we can talk?

geofjamg, 2019-02-16:
We have created a gitter chat, the link is in the README.md.

closed 2020-12-11 by MioRtia