powsybl/powsybl-core #452: Rename ExtensionProvider

geofjamg, 2018-10-07:
and ExtensionProviders as it does not provide anything (no get or provide or create method). It is very confusing and extension API is hard to understand.
It could be rename ExtensionSerializer? or maybe if we want to be more generic (probably not needed) ExtensionProcessor or ExtensionAction?

geofjamg, 2018-12-07:
Any opinion of this?

mathbagu, 2018-12-07:
I agree with your remark but not with your proposal: as this interface is also use by some "configuration loader" (see. `com.powsybl.loadflow.LoadFlowParameters` for instance). This interface is inherited by everything that provides Extension (XML or Json serializer, ConfigLoader...). It is not a provider but a base interface for this kind of classes.

Basically, this is use by the `ServiceLoader` to get plugins... Could `ExtensionLoader` be a better name?

geofjamg, 2018-12-07:
It is not a loader as there is no load method in the interface

geofjamg, 2020-01-14:
Up!
I think this bad naming is part of the weakness of the extension model. Any other idea how to name this class? 
Basically, it is the base class of serializers and loaders associated to an extension, but more generally it is so generic that it could be any processing related to an extension.

agnesLeroy, 2020-01-15:
May it be that ExtensionProvider is an ExtensionWrapper and ExtensionProviders is an ExtensionBuilder, or ExtensionLoader? Or do I get it wrong?

sylvlecl, 2020-01-15:
Maybe `ExtensionFeature` as an additional alternative?
As noted by @geofjam, subclassing this interface can be used to define any feature related to an extension. Among those features are serialization and loading from config.

Then `ExtensionProviders` is used to find or load those features, so it could be renamed `ExtensionFeatures` or `ExtensionFeaturesProvider` or `ExtensionFeaturesLoader` for example.

geofjamg, 2020-01-15:
ExtensionFeature is probably until now the best proposition

jonenst, 2020-01-20:
> Basically, this is use by the ServiceLoader to get plugins... 

I disagree. I don't see anywhere in the code where we do ServiceLoader.load(ExtentionProvider.class) . As far I as understand, this exists only to have similar methods in the interfaces related to extensions. If the methods were copied in all the interfaces, it wouldn't change anything. Maybe that's the solution, just deleting this interface ? The fact that it's hard to name is also a clue that it doesn't do much.

Just like we have a getName() method in many objects but we don't create a Nameable interface

sylvlecl, 2020-01-23:
Actually it is indirectly, [here](https://github.com/powsybl/powsybl-core/blob/8adba490db71c85bbcb753e94d36dc9c6f38e53c/commons/src/main/java/com/powsybl/commons/extensions/ExtensionProviders.java#L51). It just saves the writing of the `ExtensionProviders` class (to be renamed), which is used to find features implementations, for different types of features.

If we had common methods for nameable objects, it could be interesting to have a Nameable interface, as we have an Identifiable interface. Actually, for example, it could be interesting to have a "NameableService" interface for instance, for services that can be identified by their names.

So I would more be in favour of keeping this common base interface.

mathbagu, 2020-01-27:
Do we have an agreement? Maybe this class won’t not  be needed anymore once the refactoring will be done