powsybl/powsybl-core #556: Network XML copy does not work on node breaker network

petitrenaudseb, 2018-12-17:
See unit test on branch networkXmlCopyNodeBreaker:
in this unit test, you have two bus bar section (E1 and E2) separated by a retained switch. Consequently, you are allowed to set different values of voltage to these bus bar section (in the example 400 and 401).

When you try to copy the network, the resulting network does not keep this difference of voltage

annetill, 2019-10-21:
I am not sure to see the link with the XIIDM export issue #735. In this networkXmlCopyNodeBreaker, we have two bus bar sections separated by a retained switch. Maybe the switch has to be opened before setting two different values of voltage to the bus bar sections. In #735 , it seems there is an internal connexion between two nodes which are not mapped directly to a bus bar section. What is the link for you ?

samirromdhani, 2026-10-22:
I think the copy of the network does not keep the difference of voltage because the switch (retained) is **closed** (open="false") so the buses related to the busbarsection (E1 and E2) are of the same value on the copied network
```xml
<iidm:switch id="E_sw" name="E_sw" kind="BREAKER" retained="true" open="false" node1="0" node2="1"/>
```
But when the switch (retained) is **opened** (open="true") => the copy of the network keep the difference of voltage as expected.

```xml
<iidm:switch id="E_sw" name="E_sw" kind="BREAKER" retained="true" open="true" node1="0" node2="1"/>
```
For both cases using BusBreakerView

@petitrenaudseb, what do you think?