powsybl/powsybl-core #605: AFS listener reconnection handling

jeandemanged, 2019-02-07:
AFS remote clients who subscribe to AFS node events are notified through a WebSocket connection. 
Should this connection be broken for whatever reason, it only results in an error log on the AFS server.

We propose to enhance the RemoteListenableAppStorage and the NodeEventClient to optionally deal with reconnection. Should the connection be broken, reconnection will be re-attempted periodically

The RemoteServiceConfig will be enhanced to include parameters for the reconnection strategy:

- autoreconnect (boolean, default false)
- reconnection initial interval (integer strictly positive, default 5, in seconds)
- reconnection interval mutiplier (integer strictly positive, default 2, in seconds)
- reconnection max interval (integer strictly positive, default 3600, in seconds)

With the default parameters, there would be no reconnection attempt.
With autoreconnect set to true  and the other parameters set to default values, in the case of a disconnection, the reconnection attempts would occur after 5", then 10", 20", 40", 1'20", 2'40" .... etc incresing until a maximum of one hour interval.

mathbagu, 2019-02-07:
Your proposal seems OK. I don’t know if you have this in mind, but these parameters should be configurable in the config.yml/xml file (new module?), and maybe passed through the environment (ask @sylvlecl for details).

mathbagu, 2019-02-07:
I agree with your proposal: it’s a good idea. Do you plan to write the code yourself or not?

jeandemanged, 2022-12-13:
stale issue / closing

closed 2022-12-13 by jeandemanged