powsybl/powsybl-core #671: AFS events refactoring

geofjamg, 2019-03-05:
Only storage events can be sent. Storage events are very low level events and it would be nice if we can also send event from business layer which would be used to notify for a more high level data change.

Example: 
I want to be notified when a case is inserted in the db. Case insertion involve numbers of low level storage operations (and so on events)  and only workaround  in current library is to listen to last storage event (in that case a specific node data update). 
There are several issues with this workaround:
 - it enforces the user to know about low level implementation
 - it is not robust to implementation change
 - it is hard to understand the code
 - sometime it is not possible to find such workaround 

Proposition:
Allow business layer to sent its own event type. Event hierarchy has to be refactored and storage event will be only part of this hierarchy.
Base class of the hierarchy could be something like:
```java
interface Event {
   DateTime getTime();
   String getType();
} 
```
NodeEvent which is the  base  class of storage event will implement this interface.

Issue:
Event management is done in AppStorage (ListenableAppStorage), which cannot be the case  if business events are allowed. A complex refactoring has to be done to have an `EventStore` class used by both AppStorage implementations and AFS objects. It could be managed at AppData level.

mathbagu, 2019-03-06:
Do we really need to have the same hierarchy of events for both storage and business layers? Maybe, it's not a requirement, and the implementation could be simpler.
Another remark, do we really need an EventStore (for business layer I mean): when we will plug AFS to a broker (kafka, rabbitMQ), the message could be published when it is generated: no cache is needed here. So maybe, we can have a dummy implementation that send the message to all AppFileSystem listeners.

I think we should keep both listeners (business & storage): I don't know if you had that in mind.

geofjamg, 2019-03-06:
If we  don't have the same hierarchy of event, it means that events are just `Object`?
EventStore is only an interface that could be implemented in memory or forwarding to any kind of broker.
I was thinking about having one entry point for listeners but using a selector (corresponding to topic in case of brokers) like 'storage', 'case', etc.

geofjamg, 2023-10-11:
AFS is not part anymore of core

closed 2023-10-11 by geofjamg