powsybl/powsybl-core #672: AFS object creation consistency issue

geofjamg, 2019-03-05:
The creation of a business object like a case involve several low level operations in the storage layer like:
  - create  node
  - add data 

When can have consistency issue if another process query the parent node, between for instance node  creation and data update.

Proposition: Add a `visible` field in the node, set by default to false. Other query like children query ignore invisible nodes. Add a setVisible method in the AppStorage service to allow visibility change and so on 'business object activation' after all other update of the node is  done.

After this  evolution is done  we will be able to have such sequence of operation:
 - create node
 - add data
 - set node visibility to true

mathbagu, 2019-03-06:
I think it's a good idea to have a `visible` field in nodes, especially because it could be useful to hide some nodes, but I'm not sure we should use it to indicate that a node is not fully created.

Why don't use the creation date: when the node is fully created, the creation date is set?

geofjamg, 2019-03-06:
Creation date  is another thing not related  to node under construction.
So why not calling this attribute enable or disable ?

PR #744 (2019-06-03): Afs-Object-creation-consistency
fix #672
closed 2019-06-03 by geofjamg