powsybl/powsybl-core #678: Add convenient methods to get P, Q, V and angle

mathbagu, 2019-03-07:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current behavior?**
To get 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** via

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

geofjamg, 2019-03-08:
So what you want here is to remove terminal concept from IIDM API? because I do feel it would be a good idea to give to possibility to access same data from 2 differents places in the data model.

mathbagu, 2019-03-08:
No I just want to have convient methods on connectable to get these values. I hard to explain to someone the Terminal concept, specially to beginners with powsybl. This request has been made several times to me, Terminal concept seems not so easy to understand

geofjamg, 2019-03-08:
If terminal concept is too hard to understand we can simply remove it from the API. The worst solution for me is to keep 2 way to get the same data, it would just be a mess.
```java
network.getLine("LINE_ID").getTerminal1().getP();
network.getLine("LINE_ID").getP1();
```
So why not just keeping only second line.

geofjamg, 2019-03-08:
And also it would be consistent with xml/json serialization

annetill, 2023-10-11:
Can I close this issue ? For me, the `Terminal` object is needed to understand the sign convention. And now, for users we provide the framework in pypowsybl where we have these shortcuts.

closed 2023-10-11 by annetill