powsybl/powsybl-core #684: IIDM network model persistence

geofjamg, 2019-03-12:
* **Do you want to request a *feature* or report a *bug*?**
A feature.

* **What is the current behavior?**
To persist an IIDM network model, we have to use XML import/export. For instance in a micro-service architecture, we would need to have something like
```java
@RequestMapping(method = RequestMethod.GET, value= "...")
public ResponseEntity<String> somethingWithNetwork(...) {
    Network network = storage.read(...)
    ....
    storage.write(network)
}
```
It would be very slow in most cases because even if we need to change a few values in the network (a active power setpoint for a generator for instance) we would need to fully load and save the network (5s + 5s) espacially when there is a lot of extension.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
N/A

* **What is the expected behavior?**
N/A

* **What is the motivation / use case for changing the behavior?**
We should be able to partially load and save network. In previous case only the  generator would be loaded and re-saved. So what we need here is  kind of persistence of the network, where the network model is split and save in a database at equipment level for instance (one table per equipment, one equipment per row of the table?).
In term of API it could be somthing like:
```java
try (NetworkStore store = CassandraNetworkStore.builder()
                                         .ip("127.0.0.1)
                                         .id('AAAAAA)
                                         .build()) {
    Network network = Network.fromStore(store);
    ...
    store.save();
}
```

* **Please tell us about your environment:**
  - PowSyBl Version: 2.4.0-SNAPSHOT
  - OS Version: Linux

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

sylvlecl, 2019-07-10:
I am wondering if we should link the 2 following features :
 * Persistence of the network
 * Partial loading of some equipments of the network, or lazy loading of equipments on request

I agree that in many use cases we will need to read only a very small part of the network.
For example, we will probably have a use case for substations diagrams :
how could a REST service provide a diagram for a specific substation without loading all the network in memory ?

But does this kind of service really require to first store the whole network in database by anticipation, whereas maybe no diagram will be required, or only a few ?

Maybe we would have a specific API for partial loading, with several implementations (XML, database, ...) ?
I understand that in your proposition, the persistent characteristic would be hidden inside the Network implementation, but maybe we could expose it for specific use case, with a kind of "request" system.

geofjamg, 2019-07-10:
At some point partial loading could be implemented with XML but not as it could be with a database and a more structured data model. Partial loading means that data has to be split before being stored and the granularity of this split will allow or not efficient partial loading of the model. We already have worked in the past months of IIDM XML converter to allow split by extension and dataset (not merged this one) and that would in theory allow very raw partial loading. 

I think that will not be enough for many use cases for instance in the case of an interactive study mode where most of the interaction are open/close a switch, change a set point, change a tap so data change that involve a very small % of the data. But we still need to run a loadflow (or any other model that need almost a global read of the data), and we also need to stay efficient for this use case (but that case will be probably slower that just loading a big blob). That is the main difficulty and for me it can be solved with configurable prefetching strategies that will be set smartly depending of the use case.

I don't known if it is possible (and suitable) to share some lazy/partial loading between existing implementation and new one. That will probably be something for a second step refactoring.

geofjamg, 2023-10-11:
This feature has already been developed in the alternative IIDM implementation of network store https://github.com/powsybl/powsybl-network-store

closed 2023-10-11 by geofjamg