powsybl/powsybl-core #726: Sparse matrix builder

geofjamg, 2019-04-02:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When filling a sparse matrix, CSC underlying format impose filling columns in ascending order which can be hard to achieve. A builder could to provided to hide this complexity even if it is decrease filling performance.

* **What is the expected behavior?**

```java
MatrixFactory matrixFactory = ...
Matrix m = MatrixBuilder.newBuilder(matrixFactory, 4, 4, 3)
    .setValue(0, 0, 10)
    .addValue(1, 1, 3.4)
    ....
    .build();
```
For sparse matrix it could be implemented using a Map<Cell, Double>.
For dense matrix, builder directly forward to matrix.

* **What is the motivation / use case for changing the behavior?**
Help users to fill sparse matrix.

mathbagu, 2019-04-29:
Fixed in #715

geofjamg, 2019-04-29:
Not yet fixed

sylvlecl, 2019-04-29:
I don't think it should be closed! 
The linked merged feature does not allow to "easily" build sparse matrices. It does not allow to fill them in any order.
For example in scipy you can just create a matrix from a list of triplets (row, column, value) in any order. Values for same row and columns are added at build time.

geofjamg, 2019-05-15:
from #796 

"Still, it has not much to do with the current evolution but I am not comfortable with the this aspect of the design: SparseMatrix is actually a restriction of the general Matrix API, which makes it easy to write code which does not work for sparse implementation. It does not respect the so-called "LSP" considered as good practice.

The implementation of some method are very specific for a target use : for instance the natural implementation of the reset method would be to completely clear the array of elements, but it does not answer our need to only reset and update the elements value."

This is right and has to be considered in this PR.
A very raw proposition:
  - Matrix should be immutable
  - builder/filler class dedicated to specific way to fill a Matrix (CSC sparse, direct) and not related to Matrix implementation (for instance direct filler could  be used to created a CSC sparse matrix using intermediate map and also CSC sparse filler could  be used to fill a dense matrix).

rcourtier, 2026-10-15:
Why not create a `DenseMatrix`, fill it, and then convert it thanks to `toSparse()`?
@geofjamg

geofjamg, 2026-10-15:
Because for a large and very sparse matrix (like we have in a TSO size grid model), it would consume a lot of memory and would be also slow.

geofjamg, 2026-10-15:
What I described here is well known as a COO model which which could be then converted to a CSC or CSR model.