powsybl/powsybl-core #734: Install Script Fails on Mac

joesan, 2019-04-04:
* **Do you want to request a *feature* or report a *bug*?**

bug

* **What is the current behavior?**

As per README documentation on how to install powsybl, I should run the ./install.sh with the defaults, but I get an error message:

```
Joes-MacBook-Pro:powsybl-core joesan$ install.sh
usage: dirname path
usage: ./install.sh [options] [target...]
```

Note: I'm trying to do this on my Mac

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Steps to reproduce:

1. Get the source code
2. Navigate to the project directory
3. Run ./install.sh

* **What is the expected behavior?**

I would expect that the project is installed as specified in the README.md documentation

* **What is the motivation / use case for changing the behavior?**

I wanted to make contributions to the project!

* **Please tell us about your environment:**
  - OS Version: ... MacOS Mojave

mathbagu, 2019-04-05:
Thank you for reporting this issue. It seems that the `-f` option is not recognize on MacOS but required on linux.

As a workaround, you can probably start without the toolchain. The toolchain is useful if you want to compile optional powsybl C++ modules (MPI, KLU...). If you only need the java part (probably your case), you can try this:
```
$> mvn package
$> cp -rp distribution-core/target/powsybl <prefix>
```
The `prefix` is an existing directory specified to the toolchain using the `--prefix` option, or your home directory if it's not set.

I'm pretty sure you will have the same issue with the itools script:
```
$> cd $HOME/powsybl/bin
$> ./itools
```

In that case, you could probably fix it like this:
```
- installBinDir=$(dirname $(readlink -f $0))
+ installBinDir=$(dirname $(readlink $0))
```

joesan, 2019-04-05:
I will give it a try and post my progress! I tried to modify the script with a if else statement which can check for the OS and set the sourceDir accordingly, but that gave me some other errors and I lost track after that!

What are those C++ modules doing in the project?

mathbagu, 2019-04-05:
I will try to find a Mac to reproduce the issue and try to help you to fix it.

We don’t have many C++ modules (les than 5%). The modules relative to HPC (MPI, OpenMP) will be moved to the new dedicated powsybl-hpc repository.

There is also a small C++ module to create a JNI wrapper of the KLU from suitesparse project. This module is used by the DC loadflow implementation, in powsybl-incubator repository.

closed 2023-10-11 by geofjamg