powsybl/powsybl-core #735: Voltage/angle values lost on XIIDM export --> import for buses with no bus bar sections

sylvlecl, 2019-04-04:
* **Do you want to request a *feature* or report a *bug*?**

Ambiguous :)

* **What is the current behavior?**

In node-breaker representation of the network, voltage and angle are primarily associated to bus bar sections. In particular, when exported to XIIDM, only those voltages are exported.

It means that voltage and angle of electrical buses which do not contain any bus bar section are lost during the XIIDM export. When reimporting that file and recomputing buses from the node breaker topology, the corresponding buses will have unknown voltage and angle.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Load an XIIDM file which has 2 branches connected together through a voltage level internal connection (not a bus bar): the corresponding bus will have no associated voltage and angle value.

* **What is the expected behavior?**

I would expected that after exporting and re-importing an XIIDM file, the voltage and angle of all buses are saved and reloaded.

* **What is the motivation / use case for changing the behavior?**

Having complete load flow results in XIIDM files, to avoid having to run a new load flow or fill the holes before inspecting it safely (no NaN voltage value).

* **Please tell us about your environment:**
  - PowSyBl Version: 2.4.1

mathbagu, 2019-04-05:
This bug is related to #556

annetill, 2019-11-14:
I have an issue proposal, please give me your feedback on it. A simple way to solve this issue could be to add something like:
````
<iidm:nodeBreakerTopology .. >
      <iidm:importedBus id="?" nodes={1, 7, 15}  v=v_value angle=angle_value > 
</iidm:nodeBreakerTopology> 
````
It could be optional in the export and defined in the ````import-export-parameters-default-value```` section of the config.yml.
During the import process, we will have to check the consistency between the bus of the bus breaker view and the imported buses. All the v and angles values imported will have to be mapped in terminals, as we do for bus bar sections.
What do you think about that ?

sylvlecl, 2019-11-19:
Hi Anne,
Thank you for the proposal, it looks good.
I would propose the slightly modified version :
```xml
<iidm:nodeBreakerTopology .. >
      <iidm:bus nodes="1,7,15"  v="v_value" angle="angle_value"/> 
</iidm:nodeBreakerTopology> 
```
I don't think an ID is necessary or really useful, only the topological description is enough.
Also:
 * I think in the end it should be a plain replacement for the fields in `busBar`, not only an export option.
 * a detail but those buses should be matched with the buses of the "bus view" on import, not the "bus breaker view"

mathbagu, 2019-11-19:
Your proposal looks great. I’m not sure this is really standard to write the list of nodes like this but it should work!

jeandemanged, 2020-01-10:
It might be useful to align with CIM/CGMES here. In terms of semantics bus would rather be called TopologicalNode. 
Also making the list of TopologicalNodes non "volatile" would be useful for planning-type models, therefore managing a TopologicalNode ID could be useful e.g. for CGMES interoperability.

- node-breaker (aka 'detailled' model): TopologicalNodes are recomputed as topology changes, it is ok to discard & recreate them as needed. TopologicalNode is a set of ConnectivityNodes.
- bus-branch (aka 'planning' model): it is required to preserve topological nodes IDs on import/export. TopologicalNode is a set of Terminals.

Then as proposed the state variables voltage & angle are attached to the TopologicalNode.

From Entso-E docs:
> For a detailed substation model a topological node is a set of connectivity nodes that, in the
current network state, are connected together through any type of closed switches, including
jumpers. Topological nodes change as the current network state changes (i.e., switches, breakers,
etc. change state).
For a planning model, switch statuses are not used to form topological nodes. Instead they are
manually created or deleted in a model builder tool. Topological nodes maintained this way are
also called "busses".

sylvlecl, 2020-01-13:
I think the "bus branch" or "planning" model of CGMES is already supported in IIDM implementation, by the `BusBreakerVoltageLevel`. It relies on identifiable buses which are actually part of the model (see `ConfiguredBusImpl`), instead of being computed on demand based on an underlying topology.

Today, for those "bus breaker" voltage levels, the voltage/angle are already correctly associated to buses when exported.
But it's not the case for the "node breaker" voltage levels, for which they are associated to bus bars, hence the proposition of this evolution to export a description of "volatile" buses also in this case.

Indeed the wording is not well in line with CGMES, but changing this would be a large breaking change!
In any case it would be interesting to make this correspondance clear in the doc. It is still under work on the website: https://www.powsybl.org/docs/iidm/model/.

closed 2020-02-28 by MioRtia