powsybl/powsybl-core #862: Contingency dsl does not support expressions

nicolaslhuillier, 2019-07-10:
* **Do you want to request a *feature* or report a *bug*?**
It's rather a bug due to code refactoring

* **What is the current behavior?**
Unlike "com.powsybl.action.dsl.GroovyDslContingenciesProvider", "com.powsybl.contingency.dsl.GroovyDslContingenciesProvider" does not support expressions. Hence it is no longer possible to write things like :

for (l in network.lines) {
    if (l.terminal1.voltageLevel.nominalV >= 380.0) {
       contingency (l.id) { equipments l.id}
    }
}

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

In GroovyDslContingenciesProviderTest, just add the following in the contingency file :
def a = network.getVoltageLevel("VLGEN").nominalV >= 380.0

* **What is the expected behavior?**
We should be able to use expression (as it was the case before refactoring).
Test above should not fail.
The solution seems for ContingencyDslLoader to extends ExpressionDslLoader instead of DslLoader

* **What is the motivation / use case for changing the behavior?**
See above

* **Please tell us about your environment:**
  - PowSyBl Version: 2.5.2
  - OS Version: Linux

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

sylvlecl, 2019-07-10:
Thank you, indeed there is a bug here, but there are 2 ways to solve it :
 * Load the "expression DSL" in the contingency DSL loader, although it's not really used
 * Avoid to perform the `PowsyblAstTransformation` when the expression DSL is not loaded : I think this is more clean. The problem comes from the fact that the transformation overloads the comparison operator with a call to a `compareTo2` method which is only implemented in the expression DSL.

sylvlecl, 2019-07-10:
Or maybe a more robust variant of the first solution :
 * Load the expression DSL at the same time we apply the `PowsyblAstTransformation`, this way we enforce the use of this DSL in all DSLs, and ensure consistency between the AST and the implemented methods

mathbagu, 2019-07-10:
Fixed in #863

closed 2019-07-10 by mathbagu