powsybl/powsybl-core #885: Aliases for extensions

mathbagu, 2019-07-31:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
To keep the backward compatibility, we cannot remove deprecated extensions' serializer.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
Sometimes, if we rename an extension, it should better to allow a serializer to read both old and new versions of an extension. To do that, we need a way to associate the serializer to several extension IDs, by creating some aliases for:
- its name
- its namespace
- its prefix
In ExtensionProviders, we should also be more specific to associate an extension to its serializer. The name is probably not sufficient: the key is the complete XML namespace, but it doesn't work with JSON, as there is no namespace. I think we have to think a little bit more about this.

* **What is the motivation / use case for changing the behavior?**
We cannot refactor some extensions to make them more generic (ActivePowerControl for instance), and we have to duplicate the code to guarantee the backward compatibility.

* **Please tell us about your environment:**
  - PowSyBl Version: All
  - OS Version: All

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-01-02:
Why did you close this issue, it still seems to me to be relevant?

annetill, 2020-01-02:
It is by mistake… sorry.