powsybl/powsybl-core #915: New contingency list API

geofjamg, 2019-09-02:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
Contingency API could be improved and could be more user friendly.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
Rename `ContingenciesProvider` to `ContingencyList`
Allow better programmatic initialization:
```java
// manual list
ContingencyList l = ContingencyList.of(Contingency.line("L1"), Contingency.generator("L1"));

// automatic list
Network n = ...
ContingencyList l2 = ContingencyList.lines(n);

// automatic list with filter
ContingencyList l3 = ContingencyList.lines(n, l -> l.getTerminal1().getVoltageLevel().getNominalV() > 300);
```

* **What is the motivation / use case for changing the behavior?**
Having a more friendly contingency API.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

mathbagu, 2019-09-02:
For N-K contingencies, I propose to have something like this:
```
ContingencyList l4 = ContingencyList.of(
    Contingency.builder()
        .line("L1")
        .generator("G2")
        .build(),
    Contingency.lines("L1", "L2", "L3")
);
```

mathbagu, 2019-10-02:
It could be great to be able to load a contingencyList from a file (groovy or Json) like this:
`ContigencyList.load(file)`

Depending on the extension, it could use the relevant parser.

geofjamg, 2019-10-02:
> It could be great to be able to load a contingencyList from a file (groovy or Json) like this: `ContigencyList.load(file)`
> 
> Depending on the extension, it could use the relevant parser.

That would be indeed a nice feature. But in that case we also should be able to save it?

closed 2023-10-11 by geofjamg