powsybl/powsybl-core #964: UndirectedGraphImpl vertex assignment issue

cesar-luis-galli, 2019-10-02:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
com.powsybl.math.graph.UndirectedGraphImpl.addVertex tries to re-use previously removed vertices.  When adding a vertex (for example, increasing the node count from a NodeBreakerView) this implementation only erases the reference from removedVertices list, but it doesn't initializes the vertex, leaving it as "null".  So, when trying to add a new Switch or Connectable to this vertex the UndirectedGraphImpl.checkVertex throws PowsyblException("Vertex " + v + " not found").

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
In a node-breaker voltage level, remove all elements connected to a node, and try to re-use that node to connect new switches/connectable elements.

* **What is the expected behavior?**
After a vertex is removed (removing all elements connected to a node in a node-breaker voltage level), the vertex should be set as available to connect new elements.

* **What is the motivation / use case for changing the behaviour?**
If erasing elements leaves intermediate vertices unusable, removing and adding new elements forces to increase the node-count in node-breaker topologies unneccesarially.

* **Please tell us about your environment:**
  - PowSyBl Version:  3.0.0 SNAPSHOT 
  - OS Version: Fedora 28

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

I have tested the following fix in addVertex funcition:
    @Override
    public int addVertex() {
        int v;
        if (removedVertices.isEmpty()) {
            v = vertices.size();
            vertices.add(new Vertex<V>());
        } else {
            v = removedVertices.removeAt(0);
            //////////////////////////////////////////////////////////////////////////////////////////////
            //Fix: initialization of the previously removed vertex//////////////
            vertices.set(v, new Vertex<V>());
           ///////////////////////////////////////////////////////////////////////////////////////////////
        }
        invalidateAdjacencyList();
        notifyListener();
        return v;
    }

mathbagu, 2019-10-03:
I tried to fix this issue in this [PR](https://github.com/powsybl/powsybl-core/pull/775) few months ago.
We also created this [issue](https://github.com/powsybl/powsybl-core/issues/781) about possible API changes.

We can try your proposal: I wrote a test case for that in the PR#775. Maybe the PR#775 tried to solve too many things in the same time.

cesar-luis-galli, 2019-10-03:
> I tried to fix this issue in this [PR](https://github.com/powsybl/powsybl-core/pull/775) few months ago.
> We also created this [issue](https://github.com/powsybl/powsybl-core/issues/781) about [...]

Thanks for your reply.
Indeed, my reported issue might be considered a duplicate of this [issue](https://github.com/powsybl/powsybl-core/issues/781).

I have considered  stop calling removeIsolatedVertex it when dealing with this issue, but I wasn't sure about the impact of doing that.  It seemed to me safer to apply the fix I have proposed.

The thing is that nulling the vertex when "removing" intermediate vertices leaves those vertices unusuable, because there is no mechanism to re-initiate them again, so checkVertex will fail.
So, I think that vertices <b>shouldn't</b> go to null when removing intermediate vertices, and the ArrayList <b>should</b> shrink when removing vertices from the tail.  
In other words, besides the proposed fix, I would fix the removeVertex method too:

    @Override
    public V removeVertex(int v) {
        checkVertex(v);
        for (Edge<E> e : edges) {
            if (e != null && (e.getV1() == v || e.getV2() == v)) {
                throw new PowsyblException("An edge is connected to vertex " + v);
            }
        }
        V obj = vertices.get(v).getObject();
        if (v == vertices.size() - 1) {
            vertices.remove(v);
        } else {
            //FIX/////////////////////////////////////
            //vertices.set(v, null);
            vertices.set(v, new Vertex());
            /////////////////////////////////////////////
            removedVertices.add(v);
        }
        invalidateAdjacencyList();
        notifyListener();
        return obj;
    }

Keep in mind that NodeBreakerView.getNodeCount returns the count of <b>non-null</b> vertices, so, after removal of several nodes, this result won't be consistent with the value setted by NodeBreakerView.setNodeCount.  Avoiding nulling, this issue is avoided too, and all vertices in the range setted by NodeBreakerView are available.
Besides, another improvement to do is sorting the removedVertices ArrayList in the natural order (lower index first), so the "guess" assignment in addVertex will remove intermediate "holes" more efficiently.

mathbagu, 2019-10-03:
We have to pay attention: the nodeCount is read in the XML to allocate vertices. I think this is a bad design: we should allocate vertices when needed as I did in #775 

You're right, I think it could be a good idea to sort the unused vertices to fill holes in priority. We should also fix the `removeVertex` method to remove holes at the end of the vertices array.

cesar-luis-galli, 2019-10-03:
> 
> 
> We have to pay attention: the nodeCount is read in the XML to allocate vertices. I think this is a bad design: we should allocate vertices when needed as I did in #775

I agree.  That's why setNodeCount <b>must</b> be consistent with <b>available</b> nodes.  Otherwise, NodeBreakerView (and, consequently, XML) will fail.

> 
> You're right, I think it could be a good idea to sort the unused vertices to fill holes in priority. We should also fix the `removeVertex` method to remove holes at the end of the vertices array.

I've tested the proposed fixes I've posted before in both methods (addVertex and removeVertex), and it seems to work well (...so far).
<code>vertices</code> ArrayList shrink is "ensured" by the <code>if (v == vertices.size() - 1) {...}</code>. It will depend on the order in which nodes are removed: suppouse that the last 2 vertices are removed.  If the n-1 vertex is removed first, <code>removeVertex</> will remove only the tail vertex from <code>vertices</code>.  On the other hand, If tail node are removed sequentially, both vertices will be removed from <code>vertices</code>.  I don't really see problem with that.  Perhaps, having those tailoring unused vertices is not such a bad feature, if we have in mind that getNodeCount filters them out because they are included in <code>removedVertices</code>.
The thing is that modifying <b>removeVertex</b> to force tail vertex removal might impact on performance... (or not, it's just a thought).
So, I think the proposed fix ensures node/vertex consistency, with very low performance impact.