powsybl/powsybl-core #981: connectableBus throws exception in topology calculated from node-breaker all switches open

zamarrenolm, 2019-10-15:
* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
Exception is thrown when `BusView` of a `Terminal` is queried for `calculatedBus` in some situations.
Specifically: the voltage level has topology kind of node-breaker and all switches are opened. In this situation no bus is calculated, and the search for a `calculatedBus` fails, throwing an exception.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
A minimal example has been created as a unit test case in the branch `issue_nb_connectableBus` (module `iidm-impl`, class `NodeBreakerTest`, method `issueConnectableBus`). The following graph shows the export of the voltage level topology:

[image]

* **What is the expected behavior?**
A bus should be calculated in the voltage level. If no bus is calculated by design, then `null` could be a valid return value from the method `connectableBus`.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.0.0-SNAPSHOT
  - OS Version: *

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)
It seems the current behavior is related to change in commit 0e2578716fc913b0d8e7f779278dc5316a201d44 where the condition for throwing the exception was changed (fixed).

MioRtia, 2019-10-16:
Yes, this seems like the same issue as #939 ... What do you think @geofjamg @mathbagu, should `getConnectableBus()` be allowed to return `null` if there is no bus in the voltage level/the whole voltage level is disconnected?

geofjamg, 2019-10-16:
That would lead to a mess in existing code. 
It would be simpler to return a fictitiou bus.

zamarrenolm, 2019-10-16:
There are some points on the current code that already check for `connectableBus` returning `null`: `iidm-converter-api` `BusFilter` as an example (when the exported topology is the bus-branch).

annetill, 2019-11-08:
Can we in a first approach retruns ````null```` and wait for the possible resulting problems ? I prefer to return a ````null```` than to introduce the notion of fictitious bus. For more, the definition of a bus is that it is connected, no ?

PR #1008 (2019-11-12): BBranch calculated connectable bus from NB view can be null

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
fix #981 
fix #939

**Other information**:

(if any of the questions/checkboxes don't apply, please delete them entirely)
closed 2019-11-12 by zamarrenolm