powsybl/powsybl-core #987: AFS consistency issues and maintenance

sylvlecl, 2019-10-22:
* **Do you want to request a *feature* or report a *bug*?**

Bug.

* **What is the current behavior?**

It's easy to create inconsistencies in a filesystem tree.
For example:
```
Folder folder = afs.getRootFolder().createFolder("folder");
        folder.delete();
        Folder subFolder = folder.createFolder("subfolder");
        System.out.println(subFolder.getPath().toString());
```

The call to `getPath()` will throw an NPE when trying to go up the tree.

Once AFS is in this state and the reference to orphan nodes are lost, the API does not give any possibility to repair it, since it will be impossible to retrieve them using a path.

* **What is the expected behavior?**

There should be more safeguards stopping clients to create inconsistencies, and tools to help identify and repair inconsistencies when inconsistencies do occur.

* **Please tell us about your environment:**

PowSyBl version : v2.6.1

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

The code above has been tested with Cassandra based AFS implementation. It needs to be tested with other implementations.

sylvlecl, 2019-10-23:
In particular, the method [AppFileSystem#findProjectFolder](https://static.javadoc.io/com.powsybl/powsybl-core/2.6.1/com/powsybl/afs/AppFileSystem.html#findProjectFolder-java.lang.String-) will throw an `AfsException` if the parent project cannot be found.

pl-buiquang, 2019-12-19:
Should this safety rely on the underlying implementation of AppStorage (prevent creation of a node with an non existent parent in the exemple) ?