powsybl/powsybl-core #1032: CGMES performance regression

geofjamg, 2019-11-26:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**

On a french EHV+HV grid, import take 1m10s on my (fast) computer. Is is slow especially compare to previous performance benchmarks. I suspect recent regressions. A quick profiling shows surprising bottlenecks for instance in PropertyBag.tabultate, Logger.debug, SubstationIdMapping.grapSubstationTransformers.

[image]

Memory consumption is also huge more than 2Go. I also suspect a regression here:

[image]

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Just import a large network.

* **What is the expected behavior?**
I think, initially import was around 20s.

* **What is the motivation / use case for changing the behavior?**
To improve performance.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

MioRtia, 2019-11-26:
Hi,
Out of curiosity, when you say it initially took 20ms, was it a long time ago or only last release? I am trying to guess when the regression took place... The bottlenecks you mention are surprising to me, I don’t recall vast modifications on them.

Regarding memory, @zamarrenolm do you think it may be the refactoring of regulating controls?

geofjamg, 2019-11-26:
No it was a long time ago, probably with a very partial import, so probably it is not anymore a valid target.

MioRtia, 2019-11-26:
Okay, it would explain big increases for Logger.debug and PropertyBag.tabulate (there are way more elements to import now). It would be interesting to look at performances in not-debug mode.

The memory consumption worries me though, it shouldn't be so high. Several maps are created for regulating controls (they can only be set at the end of the conversion) but if they are the causes, we will have to think about something else.

geofjamg, 2019-11-26:
In non debug mode (you are right it is a more relevant benchmark):

[image]

geofjamg, 2019-11-26:
We can see that only SubstationIdMapping.grapSubstationTransformers seems to be suspect (1/4 of the execution time).

zamarrenolm, 2019-12-16:
I suspect that SubstationIdMapping.graphSubstationTransformers seems guilty because it is the method that provokes the initial load of cache of terminals ... I will take a more detailed look at it.

closed 2022-06-28 by geofjamg