powsybl/powsybl-core #1037: Shunt compensator import mistake

geofjamg, 2019-11-28:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Shunt compensators seem to be incorrectly imported. Taking RTE data, capacitors in Convergence become inductors in IIDM model when going through Cvg -> CGMES -> IIDM. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
Shunt compensator susceptance sign should be conserved.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2019-11-29:
Can we remove this issue ?

geofjamg, 2019-12-02:
> Can we remove this issue ?

Yes

annetill, 2019-12-02:
There is no bug in the CGMES conversion. The bug was located earlier in the workflow.

closed 2019-12-02 by annetill