powsybl/powsybl-core #1043: IEEE import

annetill, 2019-12-02:
* **Do you want to request a *feature* or report a *bug*?**

Maybe a bug.

* **What is the current behavior?**

I have analyzed the IEEE case with 14 buses. I have made an import and then run a loadflow (Hades2, https://rte-france.github.io/hades2/index.html). Some buses seems to have more precise results than others. For example, the buses with almost identical results (less than 10^-5) for voltage are Buses 1, 2, 3, 6 and 8. It corresponds to buses with generators or synchronous compensator. Buses 4, 7, 11, 12, 13 ,14 are really less precise (less than 10^-2) for voltages. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

Maybe more precise voltages for buses 4, 7, 11, 12, 13 ,14. I Wonder if we well convert in IIDM the reactances and the resistances.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

zamarrenolm, 2019-12-03:
The IEEE14 test case included in main resources for IEEE CDF model has been obtained from the Power System Test Case Archive of the University of Washington: https://labs.ece.uw.edu/pstca/pf14/pg_tca14bus.htm.

The page has a disclaimer that states: _The data has not been run through a power flow to check the answers_. 

Also, we must take into account that the voltages given in the CDF format have only 3 decimal places of precision (voltage modules are given in per-unit and angles given in degrees).

University of Illinois has published the same test case solved in different formats (not IEEE CDF). The files are available at https://icseg.iti.illinois.edu/ieee-14-bus-system/.

The IEEE14 test case given by the University of Illinois in PSS/E Raw Data format has been verified against HELM-Flow (giving the same solution). All bus balance mismatches given in the Raw data file are `< 0.01` MVA (with flows computed in HELM-Flow after import). In the Raw file, voltage modules are given with 5 decimal places in per-unit, and angles with 4 decimals in degrees.

The voltages and Q injections of regulating generators given by Illinois version of the test case can be used to build a **solved** version of the original IEEE14 test case in CDF format. This file is attached to this issue: [ieee14cdf-solved.txt](https://github.com/powsybl/powsybl-core/files/3917322/ieee14cdf-solved.txt).

The solved CDF file has been loaded in PowSyBl and in HELM-Flow, then the flows over all branches have been computed from the given voltages and resulting bus balances obtained. Results in both environments are equal with precision `< 1e-5`. 

To achieve a greatest precision in the comparison between flows computed by PowSyBl and by HELM, the IEEE CDF converter has been refactored to use the integral type `double` instead of `float` to store the floating point values (voltages, angles, impedances of branches, ...).

Using double for parameters and values in IEEE CDF model, the differences in balance mismatches computed from original voltages between PowSyBl and HELM are `< 1e-9`. The worst bus balance mismatch is 1 MVA in Q at bus 5; this is due to balances being computed from voltages rounded to the third decimal.

This verifies that the current conversion from IEEE CDF to IIDM is correct.

zamarrenolm, 2019-12-03:
As a conclusion, I suggest to use `double` instead of `float` as basic type for floating point in IEEE CDF model inside PowSyBl.

zamarrenolm, 2019-12-03:
The precision on input values (specially voltages and angles) can not be overcome as it is inherent to the file format.

geofjamg, 2019-12-03:
Thanks @zamarrenolm for your analyse! Ok for using double instead of float.

closed 2019-12-16 by geofjamg