powsybl/powsybl-core #1099: CGMES SvInjection objects define injections, not consumptions

zamarrenolm, 2020-01-03:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
According to CGMES documentation, the attribute `pInjection` of `SvInjection` objects represent _The active power injected into the bus in addition to injections from equipment terminals.  Positive sign means injection into the TopologicalNode (bus)._
In current conversion, they are mapped to IIDM `Loads`, and `pInjection`, `qInjection` are set directly as the values of attributes `P0`, `Q0` of the IIDM `Load` object, without any transformation.

* **What is the expected behavior?**
Because `P0`, `Q0` represent a consumption, and follow load sign convention, when `pInjection`, `qInjection` are used to initialize `P0`, `Q0`, a sign change must be performed:

```
Load.P0 = - SvInjection.pInjection
Load.Q0 = -SvInjection.qInjection
```

* **What is the motivation / use case for changing the behavior?**
Perform conversion following CGMES standards.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.1.0-SNAPSHOT
  - OS Version: All

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)
Be aware that there may be sample test files that do not follow CGMES definitions, and express `SvInjection` values following load sign convention (that should be considered wrong).

MioRtia, 2021-01-08:
@zamarrenolm Is it still an issue?

jeandemanged, 2026-10-14:
> 6\. I think we should fix this issue. I add it to the next release, to note that the documentation should also be updated.
>     Please feel free to challenge my idea.

Hi @colineplqt ,
I was not aware of the sign issue, good to be fixed indeed.

Note also that IIDM has a direct equivalent for cim:SvInjection: FictitiousP0/Q0 on buses (with load sign convention). We may also consider using this instead of creating fictitious loads. But this is a different issue.