powsybl/powsybl-core #1131: GroundDisconnector not supported during CGMES import

MioRtia, 2020-01-24:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
`GroundDisconnector` are ignored during CGMES import.

* **What is the expected behavior?**
`GroundDisconnector` objects should be imported as switches during CGMES conversion.

* **What is the motivation / use case for changing the behavior?**
It is in CGMES documentation and could be used.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.2.0-SNAPSHOT

* **Other information** 
It might be necessary to add another `SwitchKind` (either `GROUND_DISCONNECTOR` or `OTHER`) for switches that do not have defined type.

So-Fras, 2023-10-26:
Point of attention:
At first glance, nothing prevents someone from creating a switch with the "GROUND_DISCONNECTOR" type between 2 busbar sections. However, it does not make any sense. 
Thus, using a new type of disconnector could be seen as a weak way to model ground disconnectors.

So-Fras, 2023-10-26:
Another way to model ground disconnectors is to add a new equipment "Ground" that will be connected to the topology and a ground disconnector will just be a disconnector that connect a Ground to the topology.

Question: what kind of equipment would the "Ground" be? It does not look like an Identifiable or an Extendable. Would it be a Singleton?

NB: in the CGMES import and export, one would have to be careful when converting ground disconnectors. One object in CGMES becomes two objects in IIDM (disconnector + ground)

So-Fras, 2023-10-26:
There is another question about the loadflow calculation: what happen if we decide to close the ground disconnector?
Do we take it into account? If yes, how?

NB: in the current configuration, I think that opening or closing the ground disconnector will have no effect on the loadflow since one side of the ground disconnector is connected to "nothing".