powsybl/powsybl-core #1160: Shunt Compensator are called "shunt" on XIIDM files

MioRtia, 2020-02-17:
* **Do you want to request a *feature* or report a *bug*?**
Renaming for consistency

* **What is the current behavior?**
Shunt compensators are called `shunt` on XIIDM files

* **What is the expected behavior?**
Shunt compensators should be called `shuntCompensator` on XIIDM files.

* **What is the motivation / use case for changing the behavior?**
All other network elements are called entirely after their corresponding Java class names unless this one.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.2.0-SNAPSHOT
  - OS Version: *

annetill, 2021-01-07:
Resolved!

MioRtia, 2021-01-08:
It has not been resolved, it is still called `shunt`