powsybl/powsybl-core #1171: powsybl-cgmes-conversion test fails

redispatch, 2020-02-21:
powsybl-cgmes-conversion test fails.
See screenshot.

cloned 21.02.2020 20:00 GMT

[image]

* **Do you want to request a *feature* or report a *bug*?**
bug

mathbagu, 2020-02-21:
Could you give us more details about your environment (OS version, Java version, Maven version...).

Looking to your screenshot, the tests don’t fail, but the JVM crashes. Could you send us the dump file if it exists? Is this issue reproductible if you rerun the build?

redispatch, 2020-02-21:
It was on windows. I tried it on two different PCs. 
Maven was 3.6.3. 
JDK 13.0.2 on one PC and JDK 8 on the other PC. 
I started it directly from the command line with „mvn package“ without an IDE. 
When build with IntelliJ the error did not occur. 
The issue is reproducible with different Java versions on different PCs. It occurs again when the build is rerun.

redispatch, 2020-02-22:
OS was Win 10. 

Maybe file names get to long for windows?
I first also could not clone from git because some file pathes exceeded 260 chars. Then I changed to a shorter directory. 
I think the long path was some cgmes file. 
Just a blind guess.

mathbagu, 2020-02-24:
Hi @redispatch 
We are able to reproduce the bug. We will try to find a solution/workaround and will give you a feedback soon.

mathbagu, 2020-02-25:
The issue is linked to a timeout in maven-surefire-plugin that kills the forked JVM, due to a too long execution time. To solve this issue, we can:
- change the root level in [logback-test.xml](https://github.com/powsybl/powsybl-core/blob/master/cgmes/cgmes-conversion/src/test/resources/logback-test.xml)
- set the `forkedProcessExitTimeoutInSeconds` property to a higher value

@zamarrenolm: is it ok for you if I change the log level? Do we really need this amount of information.
@redispatch: as a workaround, you can also skip the tests

mathbagu, 2020-04-27:
The log level has been changed in #1266

closed 2020-04-27 by mathbagu