powsybl/powsybl-core #1194: Add a tool that checks that configuration file is valid

MioRtia, 2020-02-28:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When there is an error in the configuration file (a misplaced space or tab), no convenient error message is returned, some properties/modules are just not read correctly.

* **Please provide the steps to reproduce and if possible a minimal demo of the problem**
Create your yaml configuration file with incorrect indents (mix between spaces and tabs) and you will run  into issues that do not exactly state the issue

* **What is the expected behavior?**
It should be possible in some way to check that the modules/properties of a configuration file are correct.

* **What is the motivation / use case for changing the behavior?**
YAML format is prone to indent errors.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.3.0-SNAPSHOT
  - OS Version: *

rolnico, 2025-11-07:
Not sure if this tool would still be useful, there are exception thrown when the YAML file has a wrong format.
To test this, add the following test in `YamlPlatformConfigTest`:

```java
    @Test
    void testInvalidConfiguration() throws IOException {
        try (FileSystem fileSystem = Jimfs.newFileSystem(Configuration.unix())) {
            Path cfgDir = Files.createDirectory(fileSystem.getPath("config"));
            Path cfgFilePath = cfgDir.resolve("config.yml");
            try (Writer writer = Files.newBufferedWriter(cfgFilePath, StandardCharsets.UTF_8)) {
                writer.write(String.join(System.lineSeparator(),
                    "module1:",
                    "\ttab: 1",
                    "    spaces: 2"));
            }
            assertThrows(ScannerException.class, () -> new YamlModuleConfigRepository(cfgFilePath));

            try (Writer writer = Files.newBufferedWriter(cfgFilePath, StandardCharsets.UTF_8)) {
                writer.write(String.join(System.lineSeparator(),
                    "module1:",
                    "     tab: 1",
                    "    spaces: 2"));
            }
            assertThrows(ParserException.class, () -> new YamlModuleConfigRepository(cfgFilePath));

            try (Writer writer = Files.newBufferedWriter(cfgFilePath, StandardCharsets.UTF_8)) {
                writer.write(String.join(System.lineSeparator(),
                    "module1:",
                    "   tab: 1",
                    "    spaces: 2"));
            }
            assertThrows(ScannerException.class, () -> new YamlModuleConfigRepository(cfgFilePath));
        }
    }
```

colineplqt, 2026-09-09:
As @rolnico mentionned, this tool would not be useful, so I close this issue.

closed 2026-09-09 by colineplqt