powsybl/powsybl-core #1197: Standardize the CSV exports throughout POWSYBL

agnesLeroy, 2020-03-02:
* **Do you want to request a *feature* or report a *bug*?** 
This issue corresponds to a need for standardization of existing CSV exports in PowSyBl.

* **What is the current behavior?**
At the moment, the CSV exports are done with different standards: the separators can be imposed to "," or ";", or taken from the locale set of variables (care must then be taken in the tests to specify the separator, because it depends on the system's locale).

* **What is the expected behavior?**
We should provide one standard for the CSV export, to be used throughout PowSyBl.

* **What is the motivation / use case for changing the behavior?**
The aim is to simplify the use of CSV data issued by PowSyBl (when piping data in a script or importing it in Excel for instance).

* **Please tell us about your environment:**
  - PowSyBl Version: 3.2.0
  - OS Version: all supported OS

agnesLeroy, 2020-08-17:
Additional remark: at the moment the default esparator is ";" which makes the unit tests fail on Windows.

alicecaron, 2026-09-30:
Due to inactivity for more than 3 years, this issue is now closed. If you plan on solving the issue, please feel free to re-open it.

closed 2026-09-30 by alicecaron