powsybl/powsybl-core #1210: CI sometimes randomly fails at tests handling thread

MioRtia, 2020-03-09:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Sometimes, tests handling threads fail during Github Actions.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
It is random but it was the case [here](https://github.com/powsybl/powsybl-core/actions/runs/52140027).

* **What is the expected behavior?**
Multi thread tests should not randomly fail.

* **Please tell us about your environment:**
  - PowSyBl Version: 3.3.0-SNAPSHOT
  - OS Version: *

* **Other information**
@jonenst May it be connected to the [Github Action's optimization you made](https://github.com/powsybl/powsybl-core/pull/1082)? (Not sure it is related tho, it may just have been more hidden before)

mathbagu, 2020-03-09:
I notice the same behaviour and I agree. It’s probably linked to the PR we merged last week about compilation in multi thread mode

MioRtia, 2020-03-11:
Fixed by #1213

closed 2020-03-11 by MioRtia