powsybl/powsybl-core #1287: Ratio + phase tap changer useless warning

geofjamg, 2020-04-20:
* **Do you want to request a *feature* or report a *bug*?**
Don't known...

* **What is the current behavior?**
When a transformer is created with a ratio and a phase tap changer we have a warning logged:

```bash
2020-04-20_21:38:36.008 [main] WARN  c.p.i.n.i.PhaseTapChangerAdderImpl - XXXX has both Ratio and Phase Tap Changer
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
We have a lot of warning like this, when importing UCTE or CGMES files for instance. 

* **What is the expected behavior?**
I don't understand the purpose of this warning, either it is allowed and we should removed the warning either it is not and we just have to throw an exception. But "it is allowed but..." does not make any sense it just pollutes the log report.

* **What is the motivation / use case for changing the behavior?**
The purpose is just to not log useless information at least as a warning, it could be changed to trace for instance.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2020-04-21:
Hi Geoffroy. I understand your point of vue. We add this warning because the CGMES standard does not allow to have multiple RTC and PTC at the same end of a transfromer, whereas we allow it in IIDM. As we are still developping the conversion of CGMES files and we do not see exactly the problem of having multiple ratio and phase tap changers (because the integration in Hades2 and in the open loadflow is not finished). @MioRtia do you also think that a trace would be more appropriate ?

MioRtia, 2020-04-21:
@geofjamg We decided to allow it as it is "possible" in CGMES (and as a matter of fact, it happens) but it goes against the quality standards and does not really make sense for simulators (or at least, it is difficult to predict if it is consistent). However, throwing an exception might mean rejecting otherwise correct files... I know putting it as a warning is not ideal but I'm not sure putting as trace is a good idea either.

geofjamg, 2020-04-21:
So why is this warning in iidm and not cgmes side?
I have it when importing a ucte file.

MioRtia, 2020-04-21:
It is a configuration which is allowed in IIDM **but** we are not sure it really makes sense for a T3W's leg (for example) to have both a ratio and phase tap changers (in some conventions as CGMES, it is not recommended), hence the warning in IIDM when the IIDM object in this configuration is created. However, we decided against forbidding it since this configuration does exist in some files (CGMES and UCTE?) and we do not have another way to interpret such files.

geofjamg, 2020-04-21:
Ok I understand. The real problem is when regulation of both tap changer are activated. Tap changers by themselves are not a big issue (I mean just rho/alpha/rxgb deviations per tap).

closed 2021-01-08 by MioRtia