powsybl/powsybl-core #1293: Loadflow parameters and result v2

geofjamg, 2020-04-21:
* **Do you want to request a *feature* or report a *bug*?**
feature

* **What is the current behavior?**
`LoadFlowParameters` is really poor in the current version. It only have a small subset of load flow features and as a consequence all loadflow implementation rely on an extension to provide additional features. And those extensions have lots of common features...

Same for LoadFlowResult which rely on a very weak metrics Map<String, String> as a main output informations.

* **What is the expected behavior?**
Most of the parameters should move to `LoadFlowParameters` and not all implementation need to support all options. If a feature is asked but not supported by a particular impl an exception or a warning could be raised by the implementation

Based on Open LoadFlow and Hades2 here is a list of additional feature to add to LoadFlowParameters:

  - distributed slack (boolean)
  - slack bus id (in case of manual slack bus selection) 
  - balance type (PROPORTIONAL_TO_GENERATION_P, PROPORTIONAL_TO_GENERATION_P_MAX, PROPORTIONAL_TO_LOAD)
  - DC mode (boolean)
  - generator voltage remote control (boolean)
  - countries to balance (set<Country>)
  - transformer voltage control (boolean)
  - max iterations (int)
  - min and max realistic voltage (2 floats)
  - minimum reactance (SI to be compatible with different PU)
  - compute all connected components
  - ???

And to LoadFlowResult:
  - selected slack bus
  - slack bus injection
  - ???

* **What is the motivation / use case for changing the behavior?**
To facilitate load flow parameters creation.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-04-21:
@zamarrenolm do not hesitate to complete based on your HelmFlow knowledge.

murgeyseb, 2020-04-23:
I am particularly interested in the possibility to get access, through the API, to the type of computation (DC or AC). So basically I would say yes at least for this parameter.
For the rest, I have no idea if parameters from Hades and OpenLoadflow or kind of "standard" or no...
I would say that max iteration is more related to the implementation of loadflow (Newton Raphson) than to the actual loadflow business feature

annetill, 2020-04-27:
For the moment, we have in the ````LoadFlowParameters```` class:
- The initialisation of voltages and angles mode ;
- The voltage control by tap changers ;
- The regulation of phase tap changers ;
- No generation reactive limits ;
- The possibility to split shunt admittance : for the moment, this option is associated to T2W. We have an open PR to add another parameter to have a separate option for T3W. Do we really need 2 separate options ? 

About the parameters to add:
- distributed slack (boolean) and balance type (PROPORTIONAL_TO_GENERATION_P, PROPORTIONAL_TO_GENERATION_P_MAX, PROPORTIONAL_TO_LOAD) are linked in fact. does it not dangerous to have both options that can be contradictory ? 
- slack bus id (in case of manual slack bus selection) : yes but only if distributed slack is set to false : how to do that properly ? 
- DC mode (boolean) : yes.
- generator voltage remote control (boolean) : yes.
- countries to balance (set) : I am not sure that "country" is the good entity.
- transformer voltage control (boolean) : you mean remote maybe ?
- max iterations (int) : if NR based only.
- min and max realistic voltage (2 floats) : what do you mean here ?
- minimum reactance (SI to be compatible with different PU) : I am not sure it is a good idea. Finally, we don't use it in the open loadflow, do we ?
- compute all connected components : maybe, not sure. 
???

geofjamg, 2020-04-29:
>     * distributed slack (boolean) and balance type (PROPORTIONAL_TO_GENERATION_P, PROPORTIONAL_TO_GENERATION_P_MAX, PROPORTIONAL_TO_LOAD) are linked in fact. does it not dangerous to have both options that can be contradictory ?

Second is used only if distributed slack to false. It is a common way to do no?
Or we can also have only the enum with an additional value for no distributed slack.

zamarrenolm, 2020-05-05:
For me is not clear what does it mean "countries to balance". Is it related to a distributed slack alternative ? 

What would be the meaning of "DC mode" ? A DC power flow calculating only for active power ?

Computing all connected components could result in very small "chunks" of the network being evaluated. HELM-Flow allows the user to filter the size of the connected component by number of buses: if it has less than a minimum number of buses, a power flow will not be calculated unless it is the main connected component.

No problem in having additional parameters: some of them would be ignored (like max number of iterations for HELM-Flow) and some will throw an exception (HELM-Flow does not provide simplified DC power flow calculations).

sylvlecl, 2020-05-07:
I mostly agree with propositions. It should not be an issue if some options are not supported by implementations as long as it's documented. A problem is that it's difficult from our point of view to assess which options are broadly provided by loadflow implementations or not ... how could we know more about other tools options?

About proposed parameters:
 * **min/max realistic voltages** : I think this is implementation specific. It could (should?) be implemented outside of the load flow. If we still want to keep it, we will need a status in the result to identify when it happens
* **slack bus id** : can make sense to use it even when distributed slack is `true` (for example to solve max transmissible power issues linked to poor automatic choice of slack bus)

One thought about DC power flow: it could almost be a separate interface, because it's a pretty different computation. Maybe we should detail in the future what DC power flows should output: what should it do with reactive, current, etc.

PR #1481 (2020-09-28): New Loadflow parameters

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
fix #1293
closed 2020-09-28 by geofjamg