powsybl/powsybl-core #1307: Slack bus extension

geofjamg, 2020-05-10:
* **Do you want to request a *feature* or report a *bug*?**
feature

* **What is the current behavior?**
 - there is no way to configure the slack bus used in the loadflow.
 - there is no way in case of automatic slack bus selection in the loadflow to get the selected slack bus and use it for instance in the UCTE export (which requires to specify the slack bus).

* **What is the expected behavior?**

Design proposal:

IIDM extensions:
Slack bus will be marked in the IIDM network using an extension. As slack bus should be supported in node/brealer and bus/breaker voltage level, and that the way to design a topology point is very different in node/breaker and bus/break there will be 2 extensions: 
  - `NodeBreakerSlackBus` attached to a `VoltageLevel` and having a node attribute
  - `BusBreakerSlackBus` attached to a `Bus` (`ConfiguredBus`) and having no attribute

Merging:
This extension design allows nothing special to do in case of network merging. There is no need to merge extensions. The only consequence is that it is valid to have several slack bus extensions in a network resulting of the merge of several networks having already a slack bus set.

Loadflow:
It will be specified in the `LoadFlow` interface that when running a loadflow:
  - if the network contains no slack bus extensions, default selection strategy is used (like most meshed bus)
  - if only one slack bus extensions is found, it is used to locate slack bus.
  - if several slack bus extensions are found, default selection strategy is applied on all buses located from slack bus extensions.

At the end of a loadflow run:
 - if loadflow converged, all slack bus extensions are removed and a new slack bus extension is added at the selected slack bus.
 - if loadflow diverged, all slack bus extensions are let as it is.

Notice that this behavior ensure loadflow run to be idempotent. 

UCTE importer:
A slack bus extension will be added using UCTE node type information.

UCTE exporter:
If a slack bus extension is found, it is used to set UCTE node type. If no slack bus extension is found or several slack bus extensions are found, slack bus info is not exported and a warning is logged. It is not the role of UCTE export to decide what is the best slack bus.

CGMES importer:
Is there a way to get slack bus with CGMES?

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

mathbagu, 2020-05-25:
I don't know if it's a good idea (or if I don't understand well the design you propose), but it seems that both extension are not for the same type of identifiable.

For Node/Breaker voltage levels, it could be not so easy to use. For example, in UCTE exporter, we use the Bus/Breaker view to create the UCTE buses, but it will not be so easy to know if this bus is a slack bus.

I also think the propability is really low, but if there are 2 buses in a Node/Breaker voltage level that are the slack buses, you would not be able to store the information in the same voltage level?

geofjamg, 2020-05-25:
> I don't know if it's a good idea (or if I don't understand well the design you propose), but it seems that both extension are not for the same type of identifiable.
>
> For Node/Breaker voltage levels, [...]

I tried to put information to the closest Identifiable in both view, so Bus in bus/breaker view and VoltageLevel in node/breaker. So yes it means that you have to check for the right extension depending of the topology kind of the voltage level.

As an alternative with can always attach a unique extension to the VoltageLevel and the extension will have either a busId field or a node field depending of the topology kind.
I don't know if it is simpler but it is another possibility. Do you prefer this one?

geofjamg, 2020-05-25:
> I also think the propability is really low, but if there are 2 buses in a Node/Breaker voltage level that are the slack buses, you would not be able to store the information in the same voltage level?

I don't understand.There is only one slack bus per network. How could it be possible?

mathbagu, 2020-05-27:
> > I also think the propability is really low, but if there are 2 buses in a Node/Breaker voltage level that are the slack buses, you would not be able to store the information in the same voltage level?
> 
> I don't understand.There is only one slack bus per network. How could it be possible?

There is one slack node per connected components

geofjamg, 2020-05-27:
> From my understanding, there is one slack node per connected components.

Right, this is a common case. When starting a loadflow we create as many network as synchronous components and we perform independant compute on each of the network, so it is fine.
We might have the case where several extensions are attached to a synchronous component (config error or merging case) and in that case I propose to apply slack node auto selection algo to all the slack bus list to decide which one to use.

geofjamg, 2020-05-27:
> I'm not sure to have a hard recommendation for the implementation of the extensions. I like to attach the extension to the bus in the Bus/Breaker Voltage Level. But for the Node/Breaker VL, I don't know if what you propose would be easy to use. The code would be something like this :
> 
> ```
> Map<String, Set<Integer>> nodesByBus = Networks.getNodesByBus(voltageLevel);
> SlackBusExtension extension = voltageLevel.getExtension(SlackBusExtension.class);
> for (Bus bus : voltageLevel.getBusBreakerView().getBuses()) {
>     ...
>     if (nodesByBus.get(bus.getId()).contains(extension.getNode())) {
>        // I am a slack bus
>     }
> }
> ```

Yes, this is the kind of code we will need to write and indeed this is ugly.
I do not see any better solution. Any idea?

geofjamg, 2020-05-27:
> I think it's not possible to have two extension with the same name, so we would have two different extensions depending on the topology, right?

Yes

mathbagu, 2020-05-28:
If I remember well, we thought to a `TerminalRef` interface that have different implementations depending on the topology. This interface, would have a views system like the terminal. So we could imagine that:
```
public interface TerminalRef {
    public VoltageLevel getVoltageLevel();

    public class NodeBreakerView {
        public int getNode();
    }
    public class BusBreakerView {
        public Bus getBus();
    }
    public class BusView {
        public Bus getBus();
    }

    + methods to get the different views
}
```
The NodeTerminalRef and BusTerminalRef would get a Terminal from the VoltageLevel and the id of the Bus or the node, and delegate to the Terminal
```
class NodeTerminalRef implements TerminalRef {
    public class NodeBreakerViewImpl implements NodeBreakerView { ... }
    public class BusBreakerViewImpl implements BusBreakerView {
        public Bus getBus() {
            return voltageLevel.getNodeBreakerView().getTerminal(node).getBusBreakerView().getBus();
        }
    }
    public class BusViewImpl implements BusView { ... }
}
```

So the code above would be somethink like this:
```
SlackBusExtension extension = voltageLevel.getExtension(SlackBusExtension.class);
for (Bus bus : voltageLevel.getBusBreakerView().getBuses()) {
    ...
    if (extension.getTerminalRef().getBusBreakerView().getBus().getId().equals(bus.getId()) {
       // I am a slack bus
    }
}
```

If we agree to have a class to represent the reference to a terminal (for regulation purpose), we could probably use it also in this use case

geofjamg, 2020-05-28:
The problem with this approach is that it would be more natural to attach the terminal ref as a network extension and not a voltage level extension. But network level extension would also mean the need to merge extensions when merging networks (also also to have a list of terminal ref in the extension) which is something I wanted to avoid.

mathbagu, 2020-05-28:
> The problem with this approach is that it would be more natural to attach the terminal ref as a network extension and not a voltage level extension. But network level extension would also mean the [...]

I don't understand why this design change the way we attach the extension?

geofjamg, 2020-05-28:
TerminalRef already contains a voltage level id, right? So if you attach it to a voltage level you have kind of double voltage level infos. What about if you attach it to another voltage level than the one that is pointed out?

mathbagu, 2020-05-28:
You right, a TerminalRef already has info to retrieve the VoltageLevel. So I agree, it's probably not a good idea to store it in the extension, but the extension can return a terminal from its info to make the code simpler to write.

Maybe we should stop to imagine the solution as we are close enough to make a first implementation and discuss around java code instead?

annetill, 2020-06-26:
@floriand-e2r is going to try to implement a first draft solution. So it could be nice if @mathbagu or @geofjamg try to resume the solution chosen. We are going to create just one extension of the VoltageLevel but with 2 attributes depending on the topology level. Could you be more precise to describe the mecanism chosen in order not to set these 2 attributes in the same time ?

geofjamg, 2020-06-26:
I am going to update the spec with last discussions

geofjamg, 2020-06-26:
> I am going to update the spec with last discussions

Done!

mathbagu, 2020-06-26:
I would add a precision for UCTE export: it's allowed to have several slack buses extension. The slack bus in UCTE file has be in the **main synchronous** component.

MioRtia, 2021-01-08:
Resolved by #1381

closed 2021-01-08 by MioRtia