powsybl/powsybl-core #1365: Undefined permanent limit

mathbagu, 2020-06-23:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
The `CurrentLimitsAdder` accepts NaN values for permanentLimits.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
In the `Branch::isOverloaded`, we consider that a branch is overloaded if the current is greater than its permanentLimit. That relies on the fact that permanentLimits is lower of equal to temporary limit, but this is not verified.

In #1360, a fix has been done in the CGMES importer to initialize the permanentLimit is its undefined, but I wonder if this fix should be put in the IIDM implementation. This would be a breaking change, except if we initialize the permanentLimits with the lower temporaryLimit as it's done in #1360.

* **What is the expected behavior?**
Clarify if undefined permanentLimits are accepted in IIDM, and fix either the CurrentLimitsAdder or the overloading detection to be compliant to the API.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

annetill, 2021-01-07:
@MioRtia Does the PR #1359 resolves this issue?

MioRtia, 2021-01-08:
@annetill No, it does not because it breaks some compatibility on CGMES import but the code is commented in  `ValidationUtil.checkPermanentLimit`

closed 2026-02-05 by olperr1