powsybl/powsybl-core #1369: IIDM Load Model

annetill, 2020-06-26:
* **Do you want to request a *feature* or report a *bug*?**

The load injections are modeled in a very simple way in the IIDM model with a constant active injection and a constant reactive injection. First, the CGMES make a difference between conform and non conform loads. Conform loads can follow a daily pattern. In the European Merging Function, the difference is needed to distributed the slack (only on conform load). Note that the slack distribution requires the power factor of each load to remain a constant value. This means new attributes in the object and not just a type conform/non conform. 

In PSS/E model, loads can have a response that depends on voltage and on frequency. IIDM does not support this at that moment.   

* **What is the current behavior?**

We only have an constant active injection P0 and a constant active Q0.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

It is not a bug, it was just enough until today.

* **What is the expected behavior?**

At least, to have:
- For active power injection the possibility to add a percentage of the constant part to model the daily evolution ;
- For reactive power injection the possibility to add a percentage of the constant part to model the daily evolution.
This could helps to model conform load. 

For PSS/E model support:
- The possibility to add for active and reactive power injections a dependancy to (V / nominalV) ^ exp(VoltageExponent) ; Note that VoltageExponent could be different from active power to reactive power.
- A dependany to frequency that have to be defined. 

* **What is the motivation / use case for changing the behavior?**

Improve the slack distibution modeling in the open loadflow ;
Improve the PSS/E conversion.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

mathbagu, 2020-06-26:
Before implementing this, we have to remember that IIDM could be used in some use-cases where this data are not available. So we have to be really careful on the way we implement this to provide a IIDM model compatible with such use cases. In some supported format, these information doesn't exist, and it should still be possible to create Loads.

The question I have is: are they new attributes of the data model, or are they stored in an extension like the droop of the generators?

Thanks @sebalaig for the warning

geofjamg, 2020-06-28:
I think a percentage of the constant part to define to variable part does not cover common cases => in RTE internal data, the constant part is in MW/MVar.

annetill, 2020-06-30:
Thanks everyone for your answers. I think that maybe it will be enough to add new types in the ````LoadType````. To go further in the modeling, we can imagine 2 extensions, one by response modeling or just one that mixes the 2 models.

annetill, 2020-07-15:
Thanks @Miora for your PR #1377 that allow loads to have a fixed part and a variable part. This simple extension will be really useful to model slack distribution on conform load. 
Another extension is needed to model the response of load demand due to changes in the system state. As I said before, we have 2 models available in the CGMES. The second one, based on coefficients, is more spread in other industrial tools (@zamarrenolm for more details) and it seems that the first one is time-dependent. Our extensions of IIDM should remain static or not ?
After the review of IOP dataset, it seems that all cases are covered by the model with coefficients:
- A constant impedance term that is a multiple of ````Pnom * V^2```` ;
- A constant current term that is a multiple of ````Pnom * V ```` ;
- A constant power term that is a multiple of ````Pnom````.
The sum of the coefficients should be equal to ````1````.

annetill, 2023-10-11:
Solved by #2555

closed 2023-10-11 by annetill