powsybl/powsybl-core #1412: Split package

mathbagu, 2020-07-27:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
The package `com.powsybl.iidm.network.util` is present in `iidm-api` and `iidm-util`. It's not valid with Java >= 9.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
No split package:
- either we have to rename the package in one module
- either we move classes from iidm-util to iidm-api, and move the tests in iidm-tck.

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

MioRtia, 2020-08-20:
We also have the issue with `com.powsybl.iidm.network.extensions` in `iidm-api` and `iidm-extensions`.

MioRtia, 2021-01-08:
Resolved by #1575 and #1576

closed 2021-01-08 by MioRtia