powsybl/powsybl-core #1425: [MergingView] DanglingLine with empty xnode code should not be merged

mathbagu, 2020-08-07:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
If two DanglingLines have the same XNode code, they are merged together, to create a MergedLine. This is the expected behaviour. But, if two DanglingLines have an empty XNode code, they shouldn't be merged together. This attribute is optional and should be managed properly.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Create two network with DanglingLines matching by their XNode code and merge them together. If the XNode code is filled, in the mergingview, the dangling lines should have been replaced by a MergedLine. If the Xnode code is empty, the dangling lines are also replaced by a MergedLine

* **What is the expected behavior?**
In the second case, the dangling line should not be merged together and be accessible from the merging view.

* **What is the motivation / use case for changing the behavior?**
Bug fix

* **Please tell us about your environment:**
  - PowSyBl Version: 3.6.0-SNAPSHOT
  - OS Version: All

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

miovd, 2022-12-13:
Should be corrected by #2394: dangling line with null X-node only merged if dangling line with non-null X-node and same ID exists.

annetill, 2023-10-11:
Solved by #2442

closed 2023-10-11 by annetill