powsybl/powsybl-core #1537: Equipment connection/disconnection bug

geofjamg, 2020-11-03:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Algorithm used for terminal connection/disconnection is not correct and with some topology the equipment is not correctly connected/disconnected.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Here is a unit test that fails with current implementation:
```
/**
 * Copyright (c) 2020, RTE (http://www.rte-france.com)
 * This Source Code Form is subject to the terms of the Mozilla Public
 * License, v. 2.0. If a copy of the MPL was not distributed with this
 * file, You can obtain one at http://mozilla.org/MPL/2.0/.
 */
package com.powsybl.network.store.iidm.impl;

import com.powsybl.iidm.network.*;
import org.junit.Test;

import static org.junit.Assert.assertFalse;
import static org.junit.Assert.assertTrue;

/**
 *     L
 *     |
 *  ---1---
 *  |     |
 * BR1   BR2
 *  |     |
 *  ---0--- BBS1
 *
 * @author Geoffroy Jamgotchian <geoffroy.jamgotchian at rte-france.com>
 */
public class NodeBreakerDisconnectionDiamondPathBugTest {

    private Network createNetwork() {
        Network network = Network.create("test", "test");
        Substation s = network.newSubstation()
                .setId("S")
                .setCountry(Country.FR)
                .add();
        VoltageLevel vl = s.newVoltageLevel()
                .setId("VL")
                .setNominalV(400.0)
                .setTopologyKind(TopologyKind.NODE_BREAKER)
                .add();
        vl.getNodeBreakerView().newBusbarSection()
                .setId("BBS1")
                .setNode(0)
                .add();
        vl.newLoad()
                .setId("L")
                .setNode(1)
                .setP0(1)
                .setQ0(1)
                .add();
        vl.getNodeBreakerView().newBreaker()
                .setId("BR1")
                .setNode1(1)
                .setNode2(0)
                .setOpen(false)
                .add();
        vl.getNodeBreakerView().newBreaker()
                .setId("BR2")
                .setNode1(1)
                .setNode2(0)
                .setOpen(false)
                .add();
        return network;
    }

    @Test
    public void testDisconnect() {
        Network network = createNetwork();
        Load l = network.getLoad("L");
        assertTrue(l.getTerminal().isConnected());
        l.getTerminal().disconnect();
        assertFalse(l.getTerminal().isConnected());
    }
}
```

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-11-05:
Here is a link to the PR where we solved this issue in the IIDM persistent implementation with a more complexe algorithm:
https://github.com/powsybl/powsybl-network-store/pull/149
See `TerminalImpl`

flo-dup, 2021-05-06:
I can't reproduce the bug with the unit test provided (nor with the other test in [network-store NodeBreakerDisconnectionDiamondPathBugTest.java](https://github.com/powsybl/powsybl-network-store/blob/91516b99ac90ea2fb47765f632536461ce83fb93/network-store-iidm-impl/src/test/java/com/powsybl/network/store/iidm/impl/NodeBreakerDisconnectionDiamondPathBugTest.java)). Shouldn't we close the issue?