powsybl/powsybl-core #1548: Cloning network variant doesn't clone the components

mathbagu, 2020-11-26:
* **Do you want to request a *feature* or report a *bug*?**
Arguable

* **What is the current behavior?**
When a variant is created, the network variant is copied. This variant contains the connected and synchronous components but these elements are not cloned. The next time we access to the components, a connectivity calculation will be done.

It could lead to a performance issue due to this unnecessary calculation

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**
Clone also the components

* **What is the motivation / use case for changing the behavior?**
Improve the performance

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

geofjamg, 2020-11-26:
We do not clone the calculated buses either so why cloning the components? I do not see performance issue here.

mathbagu, 2020-11-26:
You right, it doesn't make sense to clone the components without cloning the buses but I was wondering if we should avoid useless connectivity calculation because cloning a variant doesn't change the connectivity. 

What is the most costly? run a connectivity calculation or clone everything?

geofjamg, 2020-11-26:
Probably cloning is a little bit faster but not so much. Not sure it worse it. Maybe we should benchmark to decide best solution.

colineplqt, 2026-09-09:
This issue is not prioritized. It can be reopen if that changes.

closed 2026-09-09 by colineplqt