powsybl/powsybl-core #1558: Make itools compatible with MacOS

mathbagu, 2020-12-04:
* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
The `itools` command doesn't work properly on MacOS because the `-f` of readlink doesn't exist (see [manpage](https://www.unix.com/man-page/osx/1/readlink/)).

Currently, we get the `bin` directory in our linux scripts using `installBinDir=$(dirname $(readlink -f $0))`. The `-f` option of readlink doesn't exist on MacOS. This options aims to follow **recursively** the symbolic links. I think the main use-cases are:
- run itools from the directory where it's installed (`./itools --help`)
- run itools specifying its path (`~/itools/bin/itools --help`)
- run itools from the PATH (`itools --help`)

I think no user have a link to the itools itself. Eventually, a user could have a link to an intermediate folder:
```
itools-v3.7.0
itools-v3.8.0
latest -> itools-v3.8.0
```
In that case, the user could run itools using `~/itools/latest/bin/itools`.

For these reasons, I propose to remove the `-f` option that should solve the issue.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Simply run `itools` on MacOS

* **What is the expected behavior?**
Be able to use powsybl on MacOS

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

mathbagu, 2020-12-04:
In the same time, I propose to fix also powsyblsh and the install.sh script (see #734)

olperr1, 2025-12-11:
The `-f` option of readlink exists since macOS 12 (released in octobre 2021).
Source: https://scriptingosx.com/2022/03/some-cli-updates-in-macos-monterey/

closed 2025-12-11 by olperr1