powsybl/powsybl-core #1610: split time series loses data

jonenst, 2021-01-28:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
 [  [ 1 2 3 ] [ 4 5 6 ] ] => split 2  => [ [ 1 2 ] [ 3 ] [ 4 ] ]
The 5,6 values are lost.

* **What is the expected behavior?**
The 5,6 values are not lost.

* **Please tell us about your environment:**
  - PowSyBl Version: all

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)
We should should what we want to return.
Possibilities:
* [[1 2][3][4][5 6]]
  but if you do that, and you split multiple timeseries at the same time with different chunks, you get a desynchronized result:
  [
  [ [ 1 2 3 ] [ 4 5 6 ] ]
  [ [ 1 2 ] [ 3 4 ] [ 5 6 ] ]
  ]
  =>
  [
  [ [ 1 2 ] [ 3 ] [ 4 ] [ 5 6 ] ]
  [ [ 1 2 ] [ 3 4 ] [ 5 6 ]
  ]

* [[1 2][3 4][5 6]]
This is the natural ideal result. But it needs to merge chunks together for this to work. the perceived complexity is probably the reason why this was not implemented.

bhorvilleur-artelys, 2021-02-01:
Hi,
I am trying to solve this problem (this being my very first contribution to PowSyBl, so please be gentle if my questions are naive).

I haven't been able to reproduce the problem. I created the following unit test:
<code java>
TimeSeriesIndex index = Mockito.mock(TimeSeriesIndex.class);
Mockito.when(index.getPointCount()).thenReturn(6);
TimeSeriesMetadata metadata = new TimeSeriesMetadata("ts1", TimeSeriesDataType.DOUBLE, Collections.emptyMap(), index);

UncompressedDoubleDataChunk chunk = new UncompressedDoubleDataChunk(0,  new double[]{1d, 2d, 3d, 4d, 5d, 6d});
DataChunk.Split<DoublePoint, DoubleDataChunk> splitChunk = chunk.splitAt(3);

StoredDoubleTimeSeries timeSeries = new StoredDoubleTimeSeries(metadata, splitChunk.getChunk1(),, splitChunk.getChunk2());
List<DoubleTimeSeries> split = timeSeries.split(2);
</code>
 
The result is : [ [1 2] [3] ] [ [4] [5 6] ].
Has this problem been solved in the meantime ? Or should I use another way of reproducing the problem ?

bhorvilleur-artelys, 2021-02-01:
That being said, I agree that it would be better if the result of the function "split" was independant on the initial display of the chunks. eg : splitting [1 2 3 4 5 6] should give the same results as splitting [[1 2 3] [4 5 6]]. I'll see what I can do about this.

commit 0f767cd6 (2021-03-01): Timeseries.split, correctly split when chunks do not perfectly align with multiple newChunkSize (#1616)

We add a method DataChunk.append to append two successive DataChunks into a new one,
and use it when chunks are not aligned with multiples of newChunkSize
fixes #1610
closed 2021-03-01 by jonenst