powsybl/powsybl-core #1644: NodeBreakerVoltageLevel connect may change the topology as unwanted side effect

zamarrenolm, 2021-03-02:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
`NodeBreakerVoltageLevel::connect` tries to connect the terminal to the closest busbar section, even if the terminal is already connected (maybe to a different busbar section).

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
As an example to demonstrate the problem with current implementation, consider a typical breaker and a half bus configuration where two lines (Line1, Line2) are connected to one busbar section (Bbs2). Busbar sections Bbs1 and Bbs2 are separated.

[image]

There could be other equipment connected to Bbs1.

Current implementation of `NodeBreakerVoltageLevel::connect` will close the path shown in orange in the diagram. This will have the effect of closing one of the unions between busbar sections, changing the topology of the substation. 

* **What is the expected behavior?**
Before closing the closest path to a busbar section, checking if the terminal is already connected (it has a `bus != null` in the `BusView`) would avoid the problem.

* **What is the motivation / use case for changing the behavior?**
When requesting the connection of a terminal, we try to keep the topology of the substation. If the substation is being operated with separated busbars and the equipment we want to connect is already connected we do not need to change the topology.

* **Please tell us about your environment:**
  - PowSyBl Version: All 
  - OS Version: N/A

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)
In a similar way, in `NodeBreakerVoltageLevel::disconnect` we could check if the terminal is already disconnected. If it is disconnected (no `bus` in the `BusView`) we do not need to explore the paths to busbar sections to open them. 

(if a question doesn't apply, you can delete it)

geofjamg, 2021-03-02:
@zamarrenolm is it the same issue as in #1537 ?
If yes, it has been fixed in the persistent IIDM impl but not yet back ported to in mem IIDM impl: https://github.com/powsybl/powsybl-network-store/blob/master/network-store-iidm-impl/src/main/java/com/powsybl/network/store/iidm/impl/TerminalImpl.java

closed 2021-05-06 by MioRtia

flo-dup, 2021-05-06:
Still unclear to me what was the issue in #1537 as I couldn't reproduce it, but that wasn't the same issue to my understanding.

PR #1720 (2021-05-06): Fix connect bug
**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?**
Yes, fixes #1644