powsybl/powsybl-core #1699: [Security analysis] On demand results for specific equipments

sylvlecl, 2021-04-26:
* **Do you want to request a *feature* or report a *bug*?**

Feature

* **What is the current behavior?**

Security analysis API only reports limit violations, which means that lots of intermediate computation results which could be interesting to the user are lost.

* **What is the expected behavior?**

It would be useful to be able to request results for specific equipments, in a similar way as sensitivity analysis.
In particular, it would be useful to request : 
* current and active power on a set of branches
* voltage on a set of buses

* **What is the motivation / use case for changing the behavior?**

Efficiently computing the post-contingency currents on a list of contingencies.

sylvlecl, 2021-05-05:
So we need to add 2 things:
1. specification of the results we want, in the input of the security analysis
2. results corresponding to those specifications, in the security analysis results

**1. Requesting some results**

The specification of results should be POJOs provided to the `SecurityAnalysisProvider::run` method.

Why POJOs: it should be serializable, and easy to provide to an external tool. Any logic (like instantiating from the network) should be performed outside of the computation

They can extend a marker interface `ResultRequest`, which could have at first an implementation `BranchResultRequest`
```java
// Specification for 1 situation
class StateResultSpecification {
   String contingencyId;       //Optional, so that it's usable for N spec, to keep it simple ...
   List<String> branchIds;    //Ask detailed information for those branches
                                           // --> should we make it more detailed? (like P, Q, I...)
   List<String> busIds;         //Ask detailed information for those buses
                                           //  --> should we make it more detailed? is it OK to use bus IDs here ?
                                           //(in particular, what about buses that change after a contingency ...
}

class SecurityAnalysisResultSpecification {
    StateResultSpecification commonResultSpecifications;
    StateResultSpecification nSituation;
    List<StateResultSpecification> contingencyResultSpecifications;
}
```

The provider should:
 * provide values of currents, active power, and reactive power on both terminals of specified branches.
 * provide voltage for all buses specified in the list of buses 

For one particular state (N-situation or contingencies), the actual specification will be the concatenation of the "common" specification and of the specific one, if it exists.

**2. Results **

geofjamg, 2021-05-17:
- In the new sensitivity API (that will come soon on core side!), we have added the concept of contingency context (https://github.com/powsybl/powsybl-open-loadflow/blob/master/src/main/java/com/powsybl/openloadflow/sensi/ContingencyContext.java). Maybe it would make sens to share it with the security analysis?

- Regarding the naming I would use "monitor" instead of "specification". I tried to search for similar concepts on google regarding simulation and data recording and it seems that monitor is widely used.

- Also I would avoid using bus id for inputs but voltage level id instead even if we end up with more results that needed.

- I agree with you that there is no need to describe that set of variable to record but just for monitored elements record all possible variables.

I would do it like this:
```java

class StateMonitor {

   ContingencyContext contingencyContext;

   List<String> branchIds;

   List<String> voltageLevelIds;
}
```

And we just pass a `List<StateMonitor>` to the run method (empty by default).

For result I am ok just maybe add voltage level id in addition to bus id as a consequence of input based of voltage level ids?
```java
class BusResult {

    String voltageLevelId;

    String busId;

    double v;

    double angle;
}
```

sylvlecl, 2021-06-18:
implemented in #1737

closed 2021-06-18 by sylvlecl