powsybl/powsybl-core #1705: IIDM slow update

geofjamg, 2021-04-28:
* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
I noticed that after a loadflow calculation, IIDM network update is very slow (line 100ms for a 6000 buses network). As a general observation every IIDM value update seem to be slow.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
To highlight the issue I just profiled an OpenLoadFlow load flow calculation and to even better hightlight the issue I changed a little but the code to update 100 times.
```diff
Index: src/main/java/com/powsybl/openloadflow/OpenLoadFlowProvider.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/src/main/java/com/powsybl/openloadflow/OpenLoadFlowProvider.java b/src/main/java/com/powsybl/openloadflow/OpenLoadFlowProvider.java
--- a/src/main/java/com/powsybl/openloadflow/OpenLoadFlowProvider.java	(revision f74a0d848d5daf245e05d24cdf90e3db655cf670)
+++ b/src/main/java/com/powsybl/openloadflow/OpenLoadFlowProvider.java	(date 1619593965640)
@@ -198,10 +198,12 @@
         for (AcLoadFlowResult result : results) {
             // update network state
             if (result.getNewtonRaphsonStatus() == NewtonRaphsonStatus.CONVERGED) {
-                result.getNetwork().updateState(!parameters.isNoGeneratorReactiveLimits(),
-                                                parameters.isWriteSlackBus(),
-                                                parameters.isPhaseShifterRegulationOn(),
-                                                parameters.isTransformerVoltageControlOn());
+                for (int i = 0; i < 100; i++) {
+                    result.getNetwork().updateState(!parameters.isNoGeneratorReactiveLimits(),
+                            parameters.isWriteSlackBus(),
+                            parameters.isPhaseShifterRegulationOn(),
+                            parameters.isTransformerVoltageControlOn());
+                }
             }
 
             LoadFlowResult.ComponentResult.Status status;
```

Profiling (in my case using JProfiler) is very clear:
[image]

We can see than, when we update the voltage magnitude of a bus (setV), almost 100% of the time is in notifyUpdate, even if there is no listener attached to the network.
 
There is 2 reasons for that:
  - AbstractConnectable::getNetwork is very slow
```java
   @Override
    public NetworkImpl getNetwork() {
        if (terminals.isEmpty()) {
            throw new PowsyblException(id + " is not attached to a network");
        }

        // During the removal of a multi terminals component (Line, 2WT or 3WT), terminals are detached from the voltage level
        return terminals.stream()
                        .map(TerminalExt::getVoltageLevel)
                        .filter(Objects::nonNull)
                        .map(VoltageLevelExt::getNetwork)
                        .findFirst()
                        .orElse(null);
    }
```
  - More generaly to get listener list we need to go back to Network by following element parent links until the network, which is slow too.  

* **What is the expected behavior?**

There is no reason this kind of code to be like 100 times faster. We should redesign the way we get listener list from every network element. 

* **What is the motivation / use case for changing the behavior?**
In the case of a load flow calculation, network update almost as time consuming as calculation itsef.

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

flo-dup, 2021-05-03:
The PR #1708 improved the update performance by a factor of 2, but this should still be investigated further.

flo-dup, 2021-05-04:
Further investigation showed that there's no quick win here, the time spent cannot be easily improved without changing iidm update model.

PR #1708 (2021-05-03): Caching the network ref in AbstractConnectable
**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** 
Yes, fixes #1705
closed 2021-05-04 by flo-dup