powsybl/powsybl-core #1725: Empty XIIDM output file when converting network from UCTE

LlucFigueras, 2021-05-09:
* **Do you want to request a *feature* or report a *bug*?**
Issue when converting network between two formats

* **What is the current behavior?**
Empty XIIDM output file when converting from a UCTE network model. The UCTE file is the IEEE8 Bus System exported from Digsilent PowerFactory.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

~/powsybl$ itools convert-network --input-file resources/UCTE/IEEE8.uct --output-format XIIDM --output-file resources/UCTE/IEEE8.xiidm
Generating file /home/lub/powsybl/resources/UCTE/IEEE8.xiidm...

The file is generated, but it is empty.

* **Please tell us about your environment:**
  - PowSyBl Version: 1.2.0 (044aaac)
  - Ubuntu 20.04

I attach both input UCTE and output XIIDM files.
[uct_xiidm_files.zip](https://github.com/powsybl/powsybl-core/files/6447939/uct_xiidm_files.zip)

mathbagu, 2021-05-10:
Hi @LlucFigueras
The UCTE file you provide is an empty grid: there is no bus, no line, no transformer and no tap change. Are you sure this is the correct file?

mathbagu, 2021-05-10:
Currently, we only support comment at the beginning of the file, that is required by the UCTE-DEF format. We do not support the comments elsewhere. To be honest, since I work with UCTE files (2008), I never see such comment blocks, but looking to the format specification, it's totally valid:
> The comment block “##C” at the top of data should describe at least the provider of the file and the reference point of the data. Additional comment blocks introduced by „##C“ and containing comment lines may be given at any place of the file. 

But it doesn't change anything to the content of your file: there is still no data in it...

In the example below, there are 3 buses (`FFFFFF11`, `XXXXXX11` and `XXXXXX12`) connected with two lines (`FFFFFF11 XXXXXX11 1` and `FFFFFF11 XXXXXX12 1`):
```
##C 2007.05.01
Generated by powsybl, 2019.09.23 17:47
Case date: 2019.09.24 19:30
##N
##ZFR
FFFFFF11              0 0        3.96708 0.00000 0.00000 0.00000
##ZXX
XXXXXX11              1 0        3.96708 0.00000 0.00000 0.00000
XXXXXX12 XNODE 2      1 0        3.96708 0.00000 0.00000 0.00000
##L
FFFFFF11 XXXXXX11 1 9 0.0000 0.2000 0.000000   8080 ABCDE
FFFFFF11 XXXXXX12 1 1 0.0000 0.2000 0.000000   5051
##T
##R
```

What does your case-file look like? Could you provide a sketch? Could you try directly with the PSS/E importer?

geofjamg, 2021-05-10:
Yes PSS/E raw file is also supported, version 32, 33 or 35.

LlucFigueras, 2021-05-10:
When converting network from RAW file to IIDM, I got the following error. The PSS/E raw file is version 33. I guess I am not doing it correctly... I haven't figured out how to. Thanks.
[image]

geofjamg, 2021-05-11:
We are going to create a 1.3.0 distribution today including the PSS/E importer.

LlucFigueras, 2021-05-12:
Many thanks! When is expected to be ready?

mathbagu, 2021-05-14:
The review of the pull request is ongoing (https://github.com/powsybl/powsybl-distribution/pull/8). I upload the XIIDM file I can get using the PSS/E converter, based on your IEEE9.raw file.
[IEEE9.zip](https://github.com/powsybl/powsybl-core/files/6477108/IEEE9.zip)

LlucFigueras, 2021-05-14:
Okay, thanks for the XIIDM file.  I only need to convert from UCTE/CIM/PSSE to XIIDM, so by now, that's fine.

MioRtia, 2021-05-17:
The 1.3.0 distribution (including PSS/E import) is up now if you want to use it!

LlucFigueras, 2021-05-17:
Thanks! However, when converting the IEEE_30_bus.raw to xiidm, it appears an error. But when converting my IEEE9.raw file to xiidm it works fine.

[image]

geofjamg, 2021-05-17:
Can you provide the IEEE_30_bus.raw ?

LlucFigueras, 2021-05-17:
[IEEE_30_bus.zip](https://github.com/powsybl/powsybl-core/files/6494743/IEEE_30_bus.zip)

Is the one already in the PSSE folder. The 57 and 118 examples files are converted successfully.

geofjamg, 2021-05-17:
Indeed import fails.
@zamarrenolm @marqueslanauja it seems we have a regression on IEEE 30 import. Any idea what could be the problem?

```java
com.powsybl.psse.model.PsseException: PSSE. Unexpected end of file

	at com.powsybl.psse.model.io.RecordGroupIOLegacyText.readRecordLine(RecordGroupIOLegacyText.java:157)
	at com.powsybl.psse.model.io.RecordGroupIOLegacyText.readRecords(RecordGroupIOLegacyText.java:142)
	at com.powsybl.psse.model.io.RecordGroupIOLegacyText.read(RecordGroupIOLegacyText.java:51)
	at com.powsybl.psse.model.io.AbstractRecordGroup.read(AbstractRecordGroup.java:132)
	at com.powsybl.psse.model.pf.io.PowerFlowRawData33.read(PowerFlowRawData33.java:62)
	at com.powsybl.psse.converter.PsseImporter.importData(PsseImporter.java:135)
	at com.powsybl.psse.converter.PsseImporterTest.importTest(PsseImporterTest.java:91)
	at com.powsybl.psse.converter.PsseImporterTest.importTest30(PsseImporterTest.java:107)
```

LlucFigueras, 2021-05-17:
By replacing the last 2 lines:

0 /END OF GNE DEVICE DATA
Q

For this ones:

0 /END OF GNE DEVICE DATA, BEGIN INDUCTION MACHINE DATA
0 /END OF INDUCTION MACHINE DATA
Q

I manage to convert it.

marqueslanauja, 2021-05-17:
The version specified in the header of the case (first line of the file) was 33 and it is wrong. The version should be 32. The `IEEE_bus_30.raw` was used in a partial test that only read the header of the file. 

PR #1736 changes the version of the case from 33 to 32 and add a complete unitary test for the case. This new test imports and exports the case.

Another way to solve the problem is what Lluc proposes since if you add at the end the induction machine data block the new file will be compatible with version 33.

geofjamg, 2021-05-17:
@marqueslanauja ok thanks for the explanation

geofjamg, 2021-05-18:
@LlucFigueras if it is ok for you we can close this issue?

LlucFigueras, 2021-05-18:
Yes, sure. Thanks