powsybl/powsybl-core #1734: Failure to import from ENTSO-E TYNDP 2020 CGMES files

bhnm-akbari, 2021-05-13:
* **Do you want to request a *feature* or report a *bug*?**
Bug/Feature

* **What is the current behavior?**
Failure to import from ENTSO-E TYNDP 2020 CGMES for specific countries.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
I would recommend running the import on the same data set. In what follows the affected countries are mentioned. Hence, it suffices to test on those countries.

* **What is the expected behavior?**
A successful import! Assumptions can be made in face of unexpected input data format. Thus, the errors could potentially be changed to warnings that declare the assumptions made.

* **Please tell us about your environment:**
  - PowSyBl Version: 1.2.0
  - OS Version: Windows 10

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

The following is a list of such errors and the files that cause the errors (Country code in parentheses):

> com.powsybl.iidm.network.ValidationException: 2 windings transformer '_603e31b8-105c-4329-880a-a52ff97914c4': step x is not set
> 20301231T235900Z_YYY_EQ_001_Updated_Updated2_Updated3_Updated4.xml (NL)
> 
> com.powsybl.iidm.network.ValidationException: Voltage level '_10771cc0-ebbd-f367-dd42-3fe964b18513': Inconsistent voltage limit range [43560.0, 242.0]
> REE_2025NT_PINT_v3_EQ_Updated_Updated2_Updated3_exported.xml (ES)
> 
> com.powsybl.iidm.network.ValidationException: Voltage level '_707de417-6edc-ec09-234b-25a6f84ff6ab': Inconsistent voltage limit range [99.001, 38.5]
> Model_ME_NT_2025_EQ_Updated_Updated2_Updated3.xml (ME)
> 
> com.powsybl.iidm.network.ValidationException: Dangling line '_d128778a-3387-11ea-9de3-b083fe720856': temporary limit value is not set
> IPTO-NT2025CY2007-7957_EQ_Updated_Updated2_Updated3.xml (GR)
> 
> com.powsybl.iidm.network.ValidationException: 2 windings transformer '_$e9e6cdfa-4ab3-4125-b5ee-ff8182b6ff28': phase regulation cannot be on if mode is FIXED
> 20241231T2300Z_www.energinet.dk_EQ_01_Updated_Updated2_Updated3.xml (DK)

mathbagu, 2021-05-14:
Powsybl aims to be used in production-ready tool where dataset exchanged should respect the format specification. In many cases, we could take assumptions that could be arguable depending on the use-case. This is the main reason why we decided to be really strict about issues in the input data. We agree that for people that are using poor quality dataset it could be frustrating.

What we can do here is to create a development branch that add the quick&dirty fixes you need to use the TYNDP dataset:
- step x is not set --> set X to 0
- Inconsistent voltage limit range [43560.0, 242.0] --> switch min and max values
- temporary limit value is not set --> skip this temporary limit ?
- phase regulation cannot be on if mode is FIXED --> set the regulation to OFF ?

Have you encountered other issues?

bhnm-akbari, 2021-05-14:
I totally see your point.

There are various other issues in the input data that PowSyBl tolerates and only warns about. Examples in reading TYNDP include the following:
> Fixed TapChanger
> Missing Terminal
> Missing Regulating control
> Fixed Permanent Limit. Reason: Several permanent limits defined for Terminal
> Ignored CurrentLimit. Reason: Defined for Equipment TwoWindingsTransformer
> Ignored CurrentLimit. Reason: Defined for Equipment ThreeWindingsTransformer.
> Fixed ... Reason: Invalid value for regulating target value. Regulation considered as local.
> Invalid Regulating Terminal. Reason: No connected IIDM terminal has been found for CGMES terminal

It could be that these issues were considered milder than the the ones that lead to error. However, none of the ones mentioned sounds fatal to me.

I agree with a development branch for this. Alternatively, you could include these dirty fixes in the main branch, but disabled by default. For the import, the user can manually turn on the fix.

Anyhow, these fixes are indeed helpful for me. Will I be able to use them in the compiled distribution with itools?

I haven't encountered any other errors. It might be possible that after solving an error, another error is encountered in a later import stage. The only thing I could do was to test different counties to compile an error list as comprehensive as possible.

mathbagu, 2021-05-14:
**DE_3PQT_v2.zip**
The version mentioned in the FullModel is not correct (TYNDP2020_NT2025_DE_v0). It's expected to be an integer ==> fixed to 0

**DK_3PQT_v1.zip**
The regulationMode of set to `FIXED` but the regulation is enabled for two windings transformer `_$e9e6cdfa-4ab3-4125-b5ee-ff8182b6ff28` ==> regulating is disabled

**ES_3PQT_v2.zip**
The high voltage limits is lower than the low voltage limits ==> voltage limits are skipped

**GR_3PQT_v1.zip**
Several temporary limits of DanglingLine `_27bc7d47-33a2-11ea-9de3-b083fe720856` have no value ==> such temporary limits are skipped

**SI_3PQT_v1.zip**
The scenarioTime is set to `N/A` where a valid date is expected ==> scenarioTime is set to the current date

**NL_3PQT_v2.zip**
3 phase tap changers have no reactance defined (_e37fad4e-2401-41b4-b66a-339ec3084a24, _f47f10d5-adf7-444e-a139-003fd51e3399, _fc3fa1bc-ac6c-4e77-9270-f64ba1c14cf8) ==> X are set to 0

All these fixes are included in this patch [powsybl-cgmes-tyndp.zip](https://github.com/powsybl/powsybl-core/files/6477779/powsybl-cgmes-tyndp.zip). You have to extract this in the `share/java` folder of your distribution to replace the `powsybl-cgmes-model-4.0.1.jar` and `powsybl-cgmes-conversion-4.0.1.jar`

With these fixes, I have been able to load all the files of the TYNDP dataset (CE and Baltics)

bhnm-akbari, 2021-05-14:
Thank you very much for the quick patch. I confirm that the TYNDP dataset can now be fully loaded. Shall I close the issue?

geofjamg, 2021-05-18:
Before closing this issue, we have to decide if we integrate to master theses fixes.
My opinion is that all of theses small fixes have to be merged.

PR #1740 (2021-05-25): Add fixes for TYNDP

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
fix #1734 

**What kind of change does this PR introduce?** *(Bug fix, feature, docs update, ...)*
Bug fixes

**What is  [... cut]
closed 2021-05-25 by MioRtia