powsybl/powsybl-core #1742: Missing data in IIDM data files after converting from raw

LlucFigueras, 2021-05-21:
* **Do you want to request a *feature* or report a *bug*?**
Feature

* **What is the current behavior?**
When converting from raw format to iidm, there is some missing data in the generators, loads, lines and transformers (p and q reference values). With small systems, I add the data manually. However, in bigger systems like the IEEE 118 bus system it is a bit time-consuming to include the data by hand. I attach a file comparing Dynawo iidm example files and the ones obtained using the psse importer.

* **What is the expected behavior?**
If possible, include p and q for loads and generators specially. Maybe it is a problem of the raw files in PSS/E, I am not sure...

* **What is the motivation / use case for changing the behavior?**
Ease of use in Dynawo simulations

* **Please tell us about your environment:**
  - PowSyBl Version: 1.3.0
  - OS Version: Ubuntu 20.04

[ComparisonIIDM.zip](https://github.com/powsybl/powsybl-core/files/6523668/ComparisonIIDM.zip)

geofjamg, 2021-05-21:
Hi,

The missing data you are pointing out, are state variables and I don't think it is contained in PSS/E raw files (because it is just "calculation" output data).
To fill them, you have to run a load flow on your case, you can do it using to "loadflow" tool, here is a link to the documentation: https://www.powsybl.org/pages/documentation/user/itools/loadflow.html

Let me know if you need help to run the load flow.

LlucFigueras, 2021-05-21:
Hi,

It is working fine, many thanks!

closed 2025-12-11 by alicecaron