powsybl/powsybl-core #1787: Observability extensions

annetill, 2021-07-09:
We have nothing in IIDM to model observability.

annetill, 2021-07-09:
What we can do:
- One extension by equipments: we should extend `Injection`, `Line`, `TwoWindingsTransformer`, `BusBarSection` and maybe in a second time `ThreeWindingsTransformer` and `ConfiguredBus`.
- In all extensions, we will have a boolean that says if the equipment is observable or not. The definition of observable regroups active power, reactive power, voltage and angle (one for all).
- In all extensions, we will have some quality information. In the `Injection` extension, we will have a `standardDeviation` for P, another one for Q and another one for V. Maybe quality information about angle is not needed. In the other extensions, we need to add the side information and it multiplies the number of field for the standard deviations. For bus bar section and configured bus, only voltage standard deviation is required.
- In all extensions, we will also have an information about redondance. The number of redondance field is similar to what we need for standard deviations.

MioRtia, 2021-07-09:
For injection, I think we can have something like that:
```xml
<io:injectionObservability observable="true">
    <io:quality standardDeviationP="0.02" standardDeviationQ="0.5" standardDeviationV="0.0" redundant="false"/>
</io:injectionObservability>
```

annetill, 2021-07-12:
> For injection, I think we can have something like that:
> 
> ```
> <io:injectionObservability observable="true">
>     <io:quality standardDeviationP="0.02" standardDeviationQ="0.5" standardDeviationV="0.0" redundant="false"/>
> </io:injectionObservability>
> ```

The `redundant` field should be multiplied for P, Q and V I think.

MioRtia, 2021-07-12:
You're right! We should have `redundantP`, `redundantQ` and so on.

MioRtia, 2021-07-12:
Should we have several `observable` fields as well?

tadam50, 2021-07-12:
> The `redundant` field should be multiplied for P, Q and V I think.

We should have something like that ?
```
<io:injectionObservability observable="true">
    <io:quality standardDeviationP="0.02" redundant="false"/>
    <io:quality standardDeviationQ="0.5" redundant="false"/>
    <io:quality standardDeviationV="0.0" redundant="false"/>
</io:injectionObservability>
```

Do we have ```quality``` fields, if ```observable``` is ```false``` ?

MioRtia, 2021-07-12:
You can have `quality` fields even if `observable` is `false`!
Maybe we should have something like this instead:
```xml
<io:injectionObservability observable="true">
    <io:qualityP standardDeviation="0.02" redundant="false"/>
    <io:qualityQ standardDeviation="0.5" redundant="false"/>
    <io:qualityV standardDeviation="0.0" redundant="false"/>
</io:injectionObservability>
```

(what do you think @annetill ?)

MioRtia, 2021-07-12:
For branches/sided objects, we could have something like:
```xml
<bo:branchObservability observable="true">
    <bo:qualityP side="ONE" standardDeviation="0.02" redundant="false"/>
    <bo:qualityP side="TWO" standardDeviation="0.02" redundant="true"/>
    <bo:qualityQ side="ONE" standardDeviation="0.5" redundant="false"/>
    <bo:qualityQ side="TWO" standardDeviation="0.78" redundant="false"/>
    <bo:qualityV side="ONE" standardDeviation="0.0" redundant="true"/>
    <bo:qualityV side="TWO" standardDeviation="0.1" redundant="false"/>
</bo:branchObservability>
```

tadam50, 2021-07-12:
> For branches/sided objects, we could have something like:
> 
> ```
> <bo:branchObservability observable="true">
>     <bo:qualityP side="ONE" standardDeviation="0.02" redundant="false"/>
>     <bo:qualityP side="TWO" standardDeviation="0.02" redundant="true"/>
>     <bo:qualityQ side="ONE" standardDeviation="0.5" redundant="false"/>
>     <bo:qualityQ side="TWO" standardDeviation="0.78" redundant="false"/>
>     <bo:qualityV side="ONE" standardDeviation="0.0" redundant="true"/>
>     <bo:qualityV side="TWO" standardDeviation="0.1" redundant="false"/>
> </bo:branchObservability>
> ```

Same methods

closed 2022-09-08 by annetill