powsybl/powsybl-core #1934: CGMES import crash when the file has the dot characters on the name

obrix, 2022-01-04:
* **Do you want to request a *feature* or report a *bug*?**

A bug

* **What is the current behavior?**

When importing a cgmes network from a zip file, if the zip file is formatted with a version number like "3.0" before the extension ".zip" then the import will crash with an error similar to : 

```
com.powsybl.cgmes.model.CgmesModelException: Listing CGMES names in data source com.powsybl.commons.datasource.ZipFileDataSource@3f6f6701
	at com.powsybl.cgmes.model.CgmesOnDataSource.names(CgmesOnDataSource.java:73)
	at com.powsybl.cgmes.model.CgmesOnDataSource.namespaces(CgmesOnDataSource.java:96)
	at com.powsybl.cgmes.model.CgmesOnDataSource.exists(CgmesOnDataSource.java:38)
	at com.powsybl.cgmes.conversion.CgmesImport.exists(CgmesImport.java:84)
	at com.powsybl.iidm.import_.Importers.findImporter(Importers.java:437)
	at com.powsybl.iidm.import_.Importers.loadNetwork(Importers.java:461)
	at com.powsybl.iidm.import_.Importers.loadNetwork(Importers.java:483)
	at com.powsybl.iidm.import_.Importers.loadNetwork(Importers.java:497)
	at com.powsybl.iidm.import_.Importers.loadNetwork(Importers.java:512)
	at com.powsybl.iidm.import_.Importers.loadNetwork(Importers.java:522)
	at com.powsybl.cgmes.conversion.CgmesImportPostProcessorTest.testImportCornet(CgmesImportPostProcessorTest.java:107)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:64)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:59)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:56)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.BlockJUnit4ClassRunner$1.evaluate(BlockJUnit4ClassRunner.java:100)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:366)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:103)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:63)
	at org.junit.runners.ParentRunner$4.run(ParentRunner.java:331)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:79)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:329)
	at org.junit.runners.ParentRunner.access$100(ParentRunner.java:66)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:293)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:413)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at com.intellij.junit4.JUnit4IdeaTestRunner.startRunnerWithArgs(JUnit4IdeaTestRunner.java:69)
	at com.intellij.rt.junit.IdeaTestRunner$Repeater.startRunnerWithArgs(IdeaTestRunner.java:33)
	at com.intellij.rt.junit.JUnitStarter.prepareStreamsAndStart(JUnitStarter.java:220)
	at com.intellij.rt.junit.JUnitStarter.main(JUnitStarter.java:53)
Caused by: java.nio.file.NoSuchFileException: D:\tmp\ENTSO-E_v3.0\RealGrid\RealGrid-Merged_v3.zip
	at java.base/sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:85)
	at java.base/sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:103)
	at java.base/sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:108)
	at java.base/sun.nio.fs.WindowsFileSystemProvider.newByteChannel(WindowsFileSystemProvider.java:235)
	at java.base/java.nio.file.Files.newByteChannel(Files.java:375)
	at java.base/java.nio.file.Files.newByteChannel(Files.java:426)
	at com.powsybl.commons.datasource.ZipFileDataSource.listNames(ZipFileDataSource.java:209)
	at com.powsybl.cgmes.model.CgmesOnDataSource.names(CgmesOnDataSource.java:69)
	... 37 more
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Import a CGMES network like this : 

`Network importedNetwork = Importers.loadNetwork("D:\\tmp\\ENTSO-E_v3.0\\RealGrid\\RealGrid-Merged_v3.0.zip");`

Then the import crash with the error above. If the zip file is renamed RealGrid-Merged.zip, then the import succeed. See the attached cgmes zip file.

* **What is the expected behavior?**

The success of the import should not depend on the name of the file.

After some debugging it seems the error may come from the class DataSourceUtil in function getBaseName at line 37.

`int pos = fileName.indexOf('.'); // find first dot in case of double extension (.xml.gz)
`

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - PowSyBl Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)
[RealGrid-Merged_v3.0.zip](https://github.com/powsybl/powsybl-core/files/7808346/RealGrid-Merged_v3.0.zip)

leblowl, 2022-01-13:
@flo-dup I could look into this if it's not already being worked on.

flo-dup, 2022-01-13:
@leblowl yes, sure you could look into this!

flo-dup, 2024-10-30:
As mentioned in PR #1943, fixed in latest release [v6.5.0](https://github.com/powsybl/powsybl-core/releases/tag/v6.5.0).

closed 2024-10-30 by flo-dup