powsybl/powsybl-core #1936: Make more IIDM extensions "variants enabled"

sylvlecl, 2022-01-10:
* **Do you want to request a *feature* or report a *bug*?**

Feature.

* **What is the current behavior?**

Some extensions carry data which "look" variant-bound, but which are actually common to all variants, see for instance `ActivePowerControlImpl`.

* **What is the expected behavior?**

The following extensions should be "variants enabled", i.e carry one value for each network variant:
 - `ActivePowerControlImpl`
 - `HvdcAngleDroopActivePowerControlImpl`
 - `HvdcOperatorActivePowerRangeImpl`
 - `CoordinatedReactiveControlImpl`
 - `RemoteReactivePowerControlImpl`
 - `VoltagePerReactivePowerControlImpl` (to be confirmed?)

* **What is the motivation / use case for changing the behavior?**

Being able to use those data in multi-variant processes.

In our actual use case, we need to generate many variant with different HVDC droop values.

CBiasuzzi, 2022-02-17:
is VoltagePerReactivePowerControlImpl, in the list, confirmed?

sylvlecl, 2022-02-17:
@annetill or @geofjamg : could you confirm it?

I think it's more something that can be adapted to the operation needs than a structural data?

sylvlecl, 2022-02-21:
@CBiasuzzi eventually, the VoltagePerReactivePowerControlImpl is NOT to be changed. Its data is more considered as constructor data, than something operational.

PR #1982 (2022-02-23): Make a set of extensions multi-variant

**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Fix #1936

**What kind of change does this PR introduce?** *(Bug fix, feature, docs update, ...)*

** [... cut]
closed 2022-02-23 by miovd