powsybl/powsybl-core #1937: Modeling fictitious injections

miovd, 2022-01-11:
* **Do you want to request a *feature* or report a *bug*?**
Feature to model fictitious injections correctly on the network (after a state estimation for example).

* **What is the current behavior?**
Fictitious injections can only be modelled as loads. However, the creation of a network component during calculation or a simulation does not allow reproducibility.

* **What is the expected behavior?**
A new component should be implemented to model fictitious injections. It should be created at bus-branch level on electrical buses.
```java
bus.setFictitiousP0(1.0).setFictitiousQ0(1.0);
```
Hence, to create it from a terminal:
```java
terminal.getBusView().getBus().setFictitiousP0(1.0).setFictitiousQ0(1.0);
```
For XML serialization, they should be written
- in node breaker as an attribute of calculated buses:
```xml
            <iidm:nodeBreakerTopology>
                <iidm:busbarSection id="D" name="E" node="0"/>
                <iidm:switch id="F" name="G" kind="DISCONNECTOR" retained="false" open="false" fictitious="true" node1="0" node2="1"/>
                <iidm:switch id="H" name="I" kind="DISCONNECTOR" retained="false" open="false" fictitious="true" node1="0" node2="3"/>
                <iidm:switch id="J" name="K" kind="BREAKER" retained="true" open="false" fictitious="true" node1="1" node2="2"/>
                <iidm:switch id="L" name="M" kind="BREAKER" retained="true" open="false" fictitious="true" node1="3" node2="4"/>
                <iidm:bus v="234.40912" angle="0.0" nodes="0,1,2,3,4" fictitiousP0="1.0" fictitiousP0="1.0"/>
            </iidm:nodeBreakerTopology>
```
- in bus-breaker and bus-branch as an attribute of buses:
```xml
            <iidm:busBreakerTopology>
                <iidm:bus id="NLOAD" ficititousP0="1.0" fictitiousQ0="1.0"/>
            </iidm:busBreakerTopology>
```
Please note that at bus-breaker level, it is written as attributes on all buses of the same electrical bus.

* **Please tell us about your environment:**
  - PowSyBl Version: 4.7.0-SNAPSHOT

annetill, 2022-01-26:
Here is my return:

After state estimation, we have the fictitious P0 and Q0 at the level of a bus from the bus view. We have to find a way to move this information to bus/breaker topology or to node/breaker topology. In the case of node/breaker topology, we must be able, getting the bus/breaker topology to retrieve the information at bus level (from the bus/breaker view).

I don't understand why we need to be able to set these P0 and Q0 from a terminal. What do you have in mind? 

To clarify:
- For bus/breaker topology, I understand that you want to put the information at calculated bus. It means that you are going to choose among several buses, maybe the first one. We have to be aware that this choice is directly linked to topology. From this view, if I want to get the fictitious injections, I have to stream over the buses?
- For node/breaker topology, the solution you are proposing is not so clear for me: you want to attach the P0 and the Q0 to an existing node, or several nodes? It means that if I want to retrieve the information, I also have to stream over the nodes of a voltage level?
- Do we need a utility method that remove all the fictitious P0 and Q0 we have introduced if the topology has changed or if we want to run again the state estimation ?

geofjamg, 2022-01-26:
> I don't understand why we need to be able to set these P0 and Q0 from a terminal. What do you have in mind?

Even if stored in a node or a configured bus, I think we should expose it in the IIDM API principally through calculated buses. Probably we should also expose it directly thought given a node or a configured bus in API but mainly for XML export/import purposes

geofjamg, 2022-01-26:
> * We have to be aware that this choice is directly linked to topology. From this view, if I want to get the fictitious injections, I have to stream over the buses?

We have to go back to how we are going to use fictitious load: mainly when creating a network for LF or any kind of simulator starting from buses, so it should be easy to take it into account by exposing it into Bus API.

geofjamg, 2022-01-26:
> * For node/breaker topology, the solution you are proposing is not so clear for me: you want to attach the P0 and the Q0 to an existing node, or several nodes?

Just one, no matter which one.

geofjamg, 2022-01-26:
> > * For node/breaker topology, the solution you are proposing is not so clear for me: you want to attach the P0 and the Q0 to an existing node, or several nodes?
> 
> Just one, no matter which one.

No because, mainly accessed though buses.

geofjamg, 2022-01-26:
> * Do we need a utility method that remove all the fictitious P0 and Q0 we have introduced if the topology has changed or if we want to run again the state estimation ?

Probably.
Also we have to check what is done, in CVG when we change the topology in a voltage level where there is a fictitious load. Is the fictitious load discarded? Or maybe it just stay arbitrary on the original node?

geofjamg, 2022-01-26:
> Probably. Also we have to check what is done, in CVG when we change the topology in a voltage level where there is a fictitious load. Is the fictitious load discarded? Or maybe it just stay [...]
In any cases, fictitious load means, we have something unknown in the modelling of the network, and when changing the topo in that area won't be perfect or accurate but there is no way to do better.

sylvlecl, 2022-01-28:
**About behaviour in Convergence tool:**
the state estimation fictitious injections are represented as loads connected to busbars:
in consequence, after topology changes, the fictitious loads remain on their bus bars.

At creation time (after state estimation), the fictitious loads are simply created on the "first" bus bar of a node, which is arbitrary.

sylvlecl, 2022-01-28:
If I understand correctly, in the end we would have :

**Storage**
* In bus breaker voltage levels, buses would store the injection values
* in node breaker voltage levels, we would have for example arrays `TDoubleArrayList` indexed on the nodes, to store the information at the node level

**API**
 * Buses would expose getters/setters for aggregated values:
```java
interface Bus {
   // Aggregates the values from underlying nodes / buses
   double getFictitiousP()
   // Resets and stores the value in underlying nodes / buses:
   // what would be the distribution logic ?
   void setFictitiousP(double value)
}
```
* We would also expose lower level access to underlying storage: 
*In-progress*

sylvlecl, 2022-01-31:
**About XML representation**
If we follow the existing list-based representation for node related data, we could use something like:
```xml
  <iidm:nodeBreakerTopology>
      ...
      <iidm:bus v="234.40912" angle="0.0" nodes="0,1,2,3,4" fictitiousP0="5.0,0,0,0,2.0" fictitiousQ0="1.0,0,0,0,0.5"/>
  </iidm:nodeBreakerTopology>
```
with required length consistency between the node related attributes.

Another option is to have node elements for nodes with injections (hopefully a few of them):
```xml
```xml
  <iidm:nodeBreakerTopology>
      ...
      <iidm:bus v="234.40912" angle="0.0" nodes="0,1,2,3,4"/>
      <iidm:node index=0 fictitiousP0="5.0" fictitiousQ0="1.0"/>
      <iidm:node index=4 fictitiousP0="2.0" fictitiousQ0="0.5"/>
  </iidm:nodeBreakerTopology>
```
The latter may be lighter overall, because only a few nodes should have fictitious injections, in the network.

miovd, 2022-02-02:
> **About XML representation** If we follow the existing list-based representation for node related data, we could use something like:
> 
> ```
>   <iidm:nodeBreakerTopology>
>       ...
>       <iidm:bus v="234.40912" angle="0.0" nodes="0,1,2,3,4" fictitiousP0="5.0,0,0,0,2.0" fictitiousQ0="1.0,0,0,0,0.5"/>
>   </iidm:nodeBreakerTopology>
> ```
> 
> with required length consistency between the node related attributes.
> 
> Another option is to have node elements for nodes with injections (hopefully a few of them):
> 
> ```
> ```xml
>   <iidm:nodeBreakerTopology>
>       ...
>       <iidm:bus v="234.40912" angle="0.0" nodes="0,1,2,3,4"/>
>       <iidm:node index=0 fictitiousP0="5.0" fictitiousQ0="1.0"/>
>       <iidm:node index=4 fictitiousP0="2.0" fictitiousQ0="0.5"/>
>   </iidm:nodeBreakerTopology>
> ```
> 
> The latter may be lighter overall, because only a few nodes should have fictitious injections, in the network.

I agree with your second serialization! 

Do you also really think fictitious injections should be settable from nodebreaker/busbreaker view? Is it okay if it is only readable at this level and settable only on calculated bus? Since they are in fact associated with bus-view bus functionnally?

sylvlecl, 2022-02-03:
I guess at least in the builder you need to be able to set it on a given node, if you want the import/export to really be idempotent.

Then for the setters I am not 100% sure, but maybe some users would want do distribute the injection in a different way than "put everything on the first node" ?

miovd, 2022-04-22:
Resolved by #2001

PR #2001 (2022-04-14): Add fictitious injections
**Please check if the PR fulfills these requirements** *(please use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes)*
- [x] The commit message follows our guidelines
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem ?** *If so, link to this issue using `'#XXX'` and skip the rest*
Fix #1937 

**Does this PR introduce a breaking change or deprecate an API?** *If yes, check the following:*
- [x]  [... cut]
closed 2022-04-22 by miovd